shadowsocks/openwrt-shadowsocks #199: 新手装ss出错了 请帮忙看看

electroun, 2016-01-06:
安装出错 请帮忙看看 多谢了
root@OpenWrt:/tmp# opkg install shadowsocks-libev-spec_2.4.3-1_ar71xx.ipk
Installing shadowsocks-libev-spec (2.4.3-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 * 	libpthread * 
 * opkg_install_cmd: Cannot install package shadowsocks-libev-spec.

aa65535, 2016-01-06:
安装之前先 `opkg update` 一下.
或者手动安装 `opkg install https://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/base/libpthread_0.9.33.2-1_ar71xx.ipk`

closed 2016-01-06 by aa65535

electroun, 2016-01-08:
恩 安装好了上面的ipk 但是安装ss的时候又出错

Configuring shadowsocks-libev-spec.
//usr/lib/opkg/info/shadowsocks-libev-spec.postinst: line 4: default_postinst: not found
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec" postinst script returned status 127.
 * opkg_configure: shadowsocks-libev-spec.postinst returned 127.

麻烦大侠看一下 如何解决 多谢 路由器是Buffalo WZR-HP-AG300H
按官网的教程刷好了https://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/openwrt-ar71xx-generic-wzr-hp-ag300h-squashfs-factory.bin

aa65535, 2016-01-08:
安装时出现 127 错误的话参考 #185 解决.

electroun, 2016-01-20:
升级到15.05 安装好所有依赖包 运行时提示出错 再请前辈指点一下

root@OpenWrt:~# /usr/bin/ss-local -c /etc/shadowsocks.json
 2016-01-20 13:42:53 INFO: initialize ciphers... aes-256-cfb
 2016-01-20 13:42:53 ERROR: bind: Address already in use
 2016-01-20 13:42:53 ERROR: Could not bind
 2016-01-20 13:42:53 ERROR: bind() error