shadowsocks/openwrt-shadowsocks #200: 关于编译的一些问题

McContax, 2016-01-07:
我的目的是安装spec版本，如果按教程gitclone到本地是不是包含了原先的libev版和spec版，还有就是听说要安装依赖包，是不是要把依赖包也加入编译进去，还有就是，按read.md来的话，有没有luci界面

aa65535, 2016-01-07:
本项目可以编译出 `原先的libev版和spec版` , 依赖包是在路由安装 ipk 的时候自动安装.
默认不带 luci , 可以到 https://github.com/aa65535/openwrt-dist-luci 下载或者自己编译.

closed 2016-01-07 by aa65535

McContax, 2016-01-09:
我使用亚马逊云搭配ubuntu编译固件，目标硬件为极路由HIWIFI HC5661A
以下是过程
sudo apt—get update
sudo apt—get upgrade
sudo apt—get install 一大堆依赖包
git clone git://git.openwrt.org/15.05/openwrt.git
sudo chmod 777 openwrt
cd ./openwrt
cp feeds.conf.default feeds.conf
./scripts/feeds update –a
./scripts/feeds install –a
make defconfig
git clone https://github.com/shadowsocks/openwrt-shadowsocks.git package/shadowsocks-libev
git clone https://github.com/aa65535/openwrt-dist-luci.git package/openwrt-dist-luci
git clone https://github.com/aa65535/openwrt-shadowvpn.git package/shadowvpn
make menuconfig<选好luci界面和shadowvpn和shadowsocks>
按以上套路配置好之后编译固件的确成功了，但是openwrt网页配置保存好之后重启路由器就又恢复原状，之前保存的配置全都重置

aa65535, 2016-01-09:
参考一下 aa65535/openwrt-shadowvpn#25

McContax, 2016-01-09:
最后我还是选择不编译用回潘多拉固件，然而在下载预编译包然后安装的时候，opkg一直验证失败，想问下有没有命令支持opkg强行安装不需要验证的

McContax, 2016-01-09:
好了，解决了安装问题，使用问题又来了，shadowsocks显示未启动怎么办，账号信息填准确了(⊙o⊙)哦