shadowsocks/openwrt-shadowsocks #207: 依然出问题

polong, 2016-01-22:
似乎依然无法运行，而且卸不掉，我现在强卸然后装回2.4.3了。
另外，安装了luci-app-shadowsocks-spec_1.3.8-1_all.ipk后似乎发现没有读取旧版的配置，弱问能让它去读取吗？不然又得重新输入。

aa65535, 2016-01-23:
请把报错信息 post 上来.

新版与旧版配置文件结构不同, 需要重新配置.

polong, 2016-01-23:
我还真不知道你指的报错信息是什么
首先，安装的信息在这里：
/tmp$ opkg install shadowsocks-libev-spec_2.4.4-1_ar71xx.ipk
Upgrading shadowsocks-libev-spec on root from 2.4.3-1 to 2.4.4-1...
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/shadowsocks is different from the conffile in the new package. The new conffile will be placed at /etc/config/shadowsocks-opkg.
Configuring shadowsocks-libev-spec.
然后安装luci1.3.8后配置里面的代理线路之后，其它项一直还是显示未找到配置
把2.4.4卸载掉出错以下错误：
[image]

aa65535, 2016-01-23:
luci 中找不到配置可以参考 #205

heyggu, 2016-01-23:
按照提示--force-remove

polong, 2016-01-24:
哦晕，原来把配置文件删除就解决问题了。

closed 2016-01-24 by aa65535