shadowsocks/openwrt-shadowsocks #211: ss co-exists with mwan3 in OperWrt router

muronghan, 2016-01-28:
严格说来可能不算是issue。。。OpenWrt中常见的第三方mwan3插件，支持根据source IP, Destination IP, port, protocal 还有ipset来分流（分头走不同的wan）。假设我上了两根ISP，国内走电信，国外走移动线路，那么mwan3只要设置两条：第一条，之前国内IP段的全部定义成一个ipset，满足这个条件的都走电信，第二条，其余的都走移动。

目前遇到的问题是，当重启Shadowsocks时，mwan3设置里的ipset的那条规则失效；如果重启mwan3，则ss代理功能失效。不知是否有望解决？

注：曾和mwan3（adze）的维护者短暂交流过，见https://forum.openwrt.org/viewtopic.php?id=39052&p=60 的1480楼。谢谢维护！

aa65535, 2016-02-06:
意思是两者不能同时开启？

muronghan, 2016-02-06:
可以同时开启，但是当ss重启后，mwan3设置的ipset不被执行（跳过？）。。。实际上我现在正是同时开启，用Chinaroute里的地址段作为mwan3的destination ip来分流，但是mwan3有个问题，就是它的destination ip段每条规则好像不能超过300条，这样，我不得不手工分拆，然后写了20几条规则。虽然目前并不影响使用，但是少了ipset，就少了很多花样，比如按域名来分流。

muronghan, 2016-02-06:
抱歉，我今天在服务器和op局端同时升级了ss-libev 2-4-5，好像这个问题已经解决了，故而close这条issue

closed 2016-02-06 by muronghan

aa65535, 2016-02-06:
可以的话，提供一下 iptables- save 的打印内容。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/211#issuecomment-180782664

muronghan, 2016-02-07:
Please see,链接: http://pan.baidu.com/s/1dDXLAF3 密码: yi2y

我的环境是OpenWrt 15.05 cc final (r46767)，mwan3 1.6.2, shadowsocks-libev-spec 2.4.5-1. 两个ISP，设置wan=移动宽带（默认），wan2=电信宽带，设置了chnroutes所有段ipset=chinaip（也给走移动的地址段设了一个ipset，叫cmcc）。

开启了OpenVPN，在mwan3里设为vpn0，作为测试（设置起来有一定难度）。所以mwan3一共有四条有效规则，依次（mwan3的特点是从上到下匹配，匹配上了不往下执行）为：

1、IP地址符合移动铁通段的，走wan（备份为wan2）;
2、IP地址符合chnroutes的，走wan2（备份为wan）;
3、某些域名走vpn0；
4、剩下的其他，通通走wan（备份为wan2）
5、其他的是缺省的，不用管，因为不会被执行。

题外话，按openvpn over shadowsocks的教材https://github.com/shadowsocks/shadowsocks/wiki/Connect-to-OpenVPN-over-Shadowsocks，目前在上面的环境中没有成功。

darknesstm, 2016-06-05:
我这实在是搞不定，mwan3只是设置了2个规则，分别是源局域网ip为xxx走wan2，目标为xxx走wan2，其他走wan， 但是用iptables nat转发符合ipset规则的流量到ss的透明代理端口都不行貌似