shadowsocks/openwrt-shadowsocks #217: How to deal with opkg's signature check?

cokebar, 2016-02-06:
I'm using your openwrt-dist url to update shadowsocks-spec with opkg. But now, there is a problem:

Downloading http://openwrt-dist.sourceforge.net/releases/ar71xx/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/openwrt_dist.
Downloading http://openwrt-dist.sourceforge.net/releases/ar71xx/packages/Packages.sig.
wget: server returned error: HTTP/1.1 404 Not Found
Signature check failed.
Remove wrong Signature file.
Downloading http://openwrt-dist.sourceforge.net/releases/luci/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/openwrt_dist_luci.
Downloading http://openwrt-dist.sourceforge.net/releases/luci/packages/Packages.sig.
wget: server returned error: HTTP/1.1 404 Not Found
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_download: Failed to download http://openwrt-dist.sourceforge.net/releases/ar71xx/packages/Packages.sig, wget returned 1.
 * opkg_download: Failed to download http://openwrt-dist.sourceforge.net/releases/luci/packages/Packages.sig, wget returned 1.

aa65535, 2016-02-06:
移除 opkg  配置里的 signature 相关的那一行即可。
我提供的源目前无法做这个签名。

closed 2016-02-06 by aa65535

cokebar, 2016-03-15:
可否增加自签名呢，opkg来更新实在方便很多
refer：
http://karl.idv.hk/2015/08/openwrt-package-signature/
貌似你只要给出公钥让我们添加即可

CzBiX, 2016-07-05:
是否有计划加入签名呢？

zixia, 2017-04-15:
@cokebar @aa65535 发布公钥 +1

CzBiX, 2017-07-19:
最终我自己动手了，写了个脚本对签名文件的请求自动生成签名，其他内容跳转回源站。

[添加我的 Public key](https://unio.czbix.com/openwrt-dist/) 之后，
把 feed 地址中的 `http://openwrt-dist.sourceforge.net/packages/` 换成 `https://unio.czbix.com/openwrt-dist/` 即可。

例如：

```
src/gz openwrt_dist https://unio.czbix.com/openwrt-dist/OpenWrt/base/ar71xx
src/gz openwrt_dist_luci https://unio.czbix.com/openwrt-dist/OpenWrt/luci
```

zixia, 2017-07-19:
@aa65535 强烈建议增加官方设置

CzBiX, 2017-07-19:
其实很简单，不知道为啥不愿意做？
我是用的系统源里的 `signify-openbsd ` 签名的。

aa65535, 2017-07-19:
已添加

```sh
wget http://openwrt-dist.sourceforge.net/packages/openwrt-dist.pub
opkg-key add openwrt-dist.pub
```