shadowsocks/openwrt-shadowsocks #230: UDP 转发不工作

psjay, 2016-03-02:
LUCI 界面显示已经运行，我配置了一个 DNS UDP 转发，dig 结果永远是失败。

closed 2016-03-02 by psjay

suikatomoki, 2016-03-08:
https://cokebar.info/archives/1053

psjay, 2016-03-08:
谢谢 @suika ，这几天查了一下的确是我 ISP 的问题，国外的 UDP 流量基本都挂了。 TCP 查询是个方法，只是我在 OpenWrt
15.05 下把 unbound 跑在 ChinaDNS 后面，怎么配置都报不支持递归查询的错误。直接 dig unbound
就没问题。另外，pdnsd 在 15.05 下已经从官方源里消失了。

dgeibi, 2016-03-08:
可以多开一个 ss-local ，让 [pcap-dnsproxy](https://github.com/wongsyrone/openwrt-Pcap_DNSProxy) 通过 TCP 查询

cokebar, 2016-03-08:
@psjay 可尝试在BB14.07里下载pdnsd 手动安装 pdnsd无内核依赖性 应该可以在CC15.05下工作（我没试过）
另一种方式 就是下载CC的Buildtool 手动从pdnsd源码编译

psjay, 2016-03-08:
用了 @dgeibi 提供的 pcap-dnsproxy 搞定了，非常感谢！

也谢谢 @cokebar 的帮助。

suikatomoki, 2016-03-09:
@cokebar 哈哈，要是ss提供tcp和udp两种dns转发选择就好了

Copox, 2016-03-12:
@dgeibi 如何多开一个?

dgeibi, 2016-03-15:
非spec 版，编辑 /etc/init.d/shadowsocks 去掉 ss-local 的注释，注意 -l 后的参数不能和 ss-redir 一样

xqdoo00o, 2016-03-21:
是长宽这 坑爹 ISP 吗? 深圳长宽 不仅封udp, tcp 443 也几分钟就 断了, 用80才行。刚开始还能用，几分钟就显示 找不到dns服务器地址了（ss-tunnel转发）

psjay, 2016-03-21:
是的，没错

dgeibi, 2016-04-09:
@psjay 发现 Pcap_DNSProxy 使用 TCP 效果不错，把 Protocol = IPv4 + UDP 改成 Protocol = IPv4 + UDP + TCP 即可，本身不用代理直接用境外 DNS 也能防污染，只不过启动略慢而已

88922493, 2016-11-30:
请问怎么实现，让vps上的SS-server同时也做为dns proxy server，  就是说国内的dns request向vps发，国外的vps-ss-server取回dns数据，再发回给国内的dns 客户端。 （在目标服务器上看到的dns request源只能是vps的ip）

flipphos, 2016-11-30:
这个可以无需ss-server。举个例子：vps和openwrt路由器上均安装pdnsd，op的dnd解析指向本地pdnsd端口，如1123，本地pdnsd再级联到vps上的非标端口，如1122，然后vps端pdnsd通过8.8.8.8查询后返回。

88922493, 2016-11-30:
具体情况是：原来国内的pc-A要dns访问 国外的server-C  （C上看到的是国内A的ip），现在如果让国内A先发dns请求给 vps上的pdnsd-server，pdnsd-server向C发请求。这种情况下C看到的是A的ip，还是vps的ip?           我想让国外的C-server看到的dns请求是vps的ip！        另外，是不是可以用iptables把dns请求发给vps的pdnsd-server，是不是国内这里就不需要用pdnsd的客户端了？

cokebar, 2017-01-06:
@88922493 你和pdnsd直接是递归查询，因此对于pdnsd的上层dns来说，客户端是pdnsd的ip，大概