shadowsocks/openwrt-shadowsocks #235: 编译时出现 zlib 错误

EdiTurn, 2016-03-08:
系统: Ubunutu 15.10 x64
SDK: OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64.tar.bz2
         OpenWrt-SDK-15.05-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2

编译输出：
#
# configuration written to .config
#
make[1]: Entering directory '/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64'
make[2]: Entering directory '/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/package/shadowsocks-libev'
rm -f /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.shadowsocks-libev.openssl_installed
(cd /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/autom4te AUTOCONF=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/autoconf AUTOMAKE=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/automake ACLOCAL=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/aclocal AUTOHEADER=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/autoheader LIBTOOLIZE=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/libtoolize LIBTOOL=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/libtool M4=/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/m4 AUTOPOINT=true /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/bin/autoreconf -v -f -i -s -B /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/host/share/aclocal -I /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/host/share/aclocal -I /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/usr/share/aclocal -I m4 -I . . || true; fi; );
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `build-aux'.
OpenWrt-libtoolize: linking file `build-aux/ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `auto'.
OpenWrt-libtoolize: linking file `auto/ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
(cd /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/./; if [ -x ./configure ]; then find /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/ -name config.guess | xargs -r chmod u+w; find /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/ -name config.guess | xargs -r -n1 cp --remove-destination /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/scripts/config.guess; find /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/ -name config.sub | xargs -r chmod u+w; find /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/ -name config.sub | xargs -r -n1 cp --remove-destination /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5:shadowsocks-libev-2.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5:shadowsocks-libev-2.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5:shadowsocks-libev-2.4.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/include/fortify -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/include " LDFLAGS="-L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/usr/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.14/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/usr/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.14/lib -znow -zrelro "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   --disable-ssp ; fi; )
configure: loading site script /tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/include/site/mips
checking for mips-openwrt-linux-gcc... mips-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether mips-openwrt-linux-musl-gcc accepts -g... yes
checking for mips-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether mips-openwrt-linux-musl-gcc understands -c and -o together... yes
[... cut ...]
Makefile:109: recipe for target '/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/build_dir/target-mips_34kc_musl-1.1.14/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/.configured_yyy' failed
make[2]: Leaving directory '/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/package/shadowsocks-libev'
package/Makefile:195: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: Leaving directory '/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64'
/tmp/OpenWrt-SDK-ar71xx-nand_gcc-5.3.0_musl-1.1.14.Linux-x86_64/include/toplevel.mk:185: recipe for target 'package/shadowsocks-libev/compile' failed

而 zlib1g zlib1g-dev 等都已经安装。

aa65535, 2016-03-10:
安装对应的软件包在 https://wiki.openwrt.org/doc/howto/buildroot.exigence 里查找。

closed 2016-03-10 by aa65535

EdiTurn, 2016-03-10:
对应的软件包是指 zlib ？已经安装了。如果有导致此问题的其他依赖，请告知一下。

EdiTurn, 2016-03-10:
已经尝试安装表格中所有软件包，然并无助于解决此问题。

EdiTurn, 2016-03-10:
@aa65535 我尝试在 DigitalOcean 上开了 CentOS、Debian 来进行编译，也出现相同的错误。

解决方法：在 SDK 根目录中运行
./scripts/feeds update -a
./scripts/feeds install zlib
./scripts/feeds install libopenssl
./scripts/feeds install libpolarssl
然后再进行编译。

hitsmaxft, 2016-04-09:
更新base部分就足够了  `./scripts/feeds update base`
```
./scripts/feeds install zlib
./scripts/feeds install libopenssl
./scripts/feeds install libpolarssl
``
然后安装依赖

另外 openwrt的编译脚本比较坑人, 依赖 svn (subversion) , 执行失败也不提示, 还是shell 一步一步debug 才找到的

vimagick, 2016-08-02:
You can edit [package/shadowsocks-libev/Makefile][1], append `+zlib` as dependency.

```diff
- Package/shadowsocks-libev-spec = $(call Package/shadowsocks-libev/Default,openssl,(OpenSSL),+libopenssl +libpthread +ipset +ip)
+ Package/shadowsocks-libev-spec = $(call Package/shadowsocks-libev/Default,openssl,(OpenSSL),+libopenssl +libpthread +ipset +ip +zlib)
```

[1]: https://github.com/shadowsocks/openwrt-shadowsocks/blob/master/Makefile#L42