shadowsocks/openwrt-shadowsocks #236: pptp拨号进来的ip,不能用透明代理

thankeyes, 2016-03-15:
路由开了pptpd服务,pptp拨号进来的ip,不能用透明代理,已经把ip段添加进去luci了,还是不行,
有解决办法吗?

cokebar, 2016-03-16:
ss-spec只作用与防火墙区域“LAN”。尝试将vpn的interface加入LAN区域看是否可行

thankeyes, 2016-03-16:
之前改过bin里的rule里某个规则,可以实现非LAN也能用到透明代理,后面没有保存,路由reset后我又找不回来了.昨晚找到一个解决办法,自己添加一段iptable

iptables -t nat -N shadowsocks
# Ignore your shadowsocks server's addresses
# It's very IMPORTANT, just be careful.
iptables -t nat -A shadowsocks -d XXXXXXXX -j RETURN
# Ignore LANs IP address
iptables -t nat -A shadowsocks -d 0.0.0.0/8 -j RETURN
iptables -t nat -A shadowsocks -d 10.0.0.0/8 -j RETURN
iptables -t nat -A shadowsocks -d 127.0.0.0/8 -j RETURN
iptables -t nat -A shadowsocks -d 169.254.0.0/16 -j RETURN
iptables -t nat -A shadowsocks -d 172.16.0.0/12 -j RETURN
iptables -t nat -A shadowsocks -d 224.0.0.0/4 -j RETURN
iptables -t nat -A shadowsocks -d 240.0.0.0/4 -j RETURN
###zidingyi
iptables -t nat -A shadowsocks -d 192.168.0.0/16 -j RETURN
#guonei ip
iptables -t nat -A shadowsocks -d 14.0.12.0/255.255.252.0 -j RETURN
iptables -t nat -A shadowsocks -d 14.1.0.0/255.255.252.0 -j RETURN
iptables -t nat -A shadowsocks -d 14.192.60.0/255.255.252.0 -j RETURN
iptables -t nat -A shadowsocks -d 14.192.64.0/255.255.224.0 -j RETURN
iptables -t nat -A shadowsocks -d 27.0.0.0/255.192.0.0 -j RETURN
iptables -t nat -A shadowsocks -d 27.50.40.0/255.255.248.0 -j RETURN
#zixing tianjia guonei ip
# Anything else should be redirected to shadowsocks's local port
#zidingyi neiwang ipdizhi huo ipduan
iptables -t nat -A shadowsocks -s 192.168.11.0/24  -p tcp -j REDIRECT --to-ports XXXX
iptables -t nat -A PREROUTING -p tcp -j shadowsocks

aa65535, 2016-04-03:
#78

closed 2016-04-03 by aa65535