shadowsocks/openwrt-shadowsocks #238: 能提供ss-redir、ss-rules、ss-tunnel各种参数的默认值吗？

Kumhy, 2016-03-19:
例如 ss-redir 的 -n Max number of open files 如果启动的时候不带这个参数默认值是多少。

最好能详细讲解下 ss-rules 各种参数
`-i <ip_list_file>`
`-b <wan_ips>`
这两个都忽略不使用，而只使用
`-w <wan_ips>`

实际上我想实现的是只有gfwlist内的走ss

cokebar, 2016-03-19:
只有gfwlist的走ss不适用于ss-rules，完全相反的两种方案，一个白名单、一个黑名单，完全不兼容

Kumhy, 2016-03-19:
我看参数说明有
`-w <wan_ips>    wan ip of will be forwarded` 强制走代理的IP
如果 -i -b都不工作，这样是不是就可以实现呢？

s7lx, 2016-03-30:
@cokebar 我实现了一半的这个规则。先让-w 加入两条，就是0.0.0.0/1 和128.0.0.1/1。这样就关闭了黑名单，然后单独写一条
iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
iptables -t nat -A OUTPUT -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
就好了
这里gfwlist是一个ipset

cokebar, 2016-03-31:
@s7lx 这样你直接用原版就行了不用spec版 ss-rules等于是废的
当然你也可以考虑整合一下两种方案 做个带切换两种方案的版本出来

Kumhy, 2016-03-31:
只有gfwlist走ss完整简单的方案谁有啊

closed 2016-03-31 by Kumhy

cokebar, 2016-03-31:
@Kumhy https://cokebar.info/archives/962