shadowsocks/openwrt-shadowsocks #245: 新版极路由2 依赖iptables-mod-tproxy, 但是opkg 没有iptables-mod-tproxy

GreatFireWall, 2016-04-02:
root@Hiwifi:~# opkg install shadowsocks-libev-spec_2.4.5-1_ramips_24kec.ipk 
Installing shadowsocks-libev-spec (2.4.5-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 * 	iptables-mod-tproxy * 
 * opkg_install_cmd: Cannot install package shadowsocks-libev-spec.

closed 2016-04-03 by aa65535

blog2i2j, 2016-05-11:
不安装这个, 在luci中会显示不支持udp服务器啊, 是需要自己另外安装吗?

aa65535, 2016-05-11:
@blog2i2j 可以自己手动安装，安装之后支持设置UDP服务器
```
opkg update
opkg install iptables-mod-tproxy
```

shihhuangti, 2016-10-20:
我安装了以后还是不支持UDP relay server，需要手工配置什么？

lifong2002, 2016-11-07:
我也是，安装了iptables-mod-tproxy还是提示不可用。
顺便问一下安装过程中提示的
Collected errors:
 * pkg_run_script: package "shadowsocks-libev" postinst script returned status 127.
 * opkg_configure: shadowsocks-libev.postinst returned 127.
是什么问题？我看别人说的可以不管，但是还是想学习学习。

commit 4989d9f3 (2016-04-03): remove dependency of iptables-mod-tproxy

closed #245