shadowsocks/openwrt-shadowsocks #247: .脸书，谷歌能正常打开的情况下，但youtube不一定能打开！

oe77, 2016-04-08:
固件：OpenWrt Designated Driver r49119
shadowsocks服务端：最新的ShadowsocksR/Shadowsocks libev
发现问题：
1.脸书，谷歌能正常打开的情况下，但youtube不一定能打开！过不知多久又正常了！
2.正在观看youtube的情况下，缓存会一直进行，但不能打开新的youtube页面！过不知多久又正常了！
尝试转换非本人自编的固件，但问题还是会出现以上问题，但在shadowsocks客户端并不会出现以上问题！！求解释

Fri Apr  8 13:38:02 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:03 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:04 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:04 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:04 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:04 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 daemon.warn pppd[1029]: Timeout waiting for PADO packets
Fri Apr  8 13:38:05 2016 daemon.err pppd[1029]: Unable to complete PPPoE Discovery
Fri Apr  8 13:38:05 2016 daemon.info pppd[1029]: Exit.
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:05 2016 daemon.notice netifd: Interface 'wan' is now down
Fri Apr  8 13:38:05 2016 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr  8 13:38:06 2016 daemon.info pppd[1267]: Plugin rp-pppoe.so loaded.
Fri Apr  8 13:38:06 2016 daemon.info pppd[1267]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Fri Apr  8 13:38:06 2016 daemon.notice pppd[1267]: pppd 2.4.7 started by root, uid 0
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:06 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:07 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:08 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:09 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:09 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:09 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:09 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:09 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
Fri Apr  8 13:38:10 2016 kern.err /usr/bin/ss-tunnel[1121]: [udp] sendto_remote: Network unreachable
[... 99 lines cut ...]
Fri Apr  8 13:38:31 2016 kern.notice kernel: [   31.192069] random: nonblocking pool is initialized
Fri Apr  8 13:38:32 2016 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1173]: exiting on receipt of SIGTERM
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: started, version 2.75 cachesize 150
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack ipset no-auth no-DNSSEC loop-detect inotify
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: DNS service limited to local subnets
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq-dhcp[1425]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: using nameserver 115.159.157.26#53
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: using nameserver 127.0.0.1#1053
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: using local addresses only for domain lan
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: read /etc/hosts - 4 addresses
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/dhcp - 2 addresses
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/odhcpd - 1 addresses
Fri Apr  8 13:38:32 2016 daemon.info dnsmasq-dhcp[1425]: read /etc/ethers - 0 addresses
Fri Apr  8 13:38:39 2016 daemon.warn odhcpd[844]: DHCPV6 SOLICIT IA_NA from 000100011e75e289086266c6f0e6 on br-lan: ok fdb7:90c5:22a1::927/128 
Fri Apr  8 13:38:39 2016 daemon.warn odhcpd[844]: DHCPV6 REQUEST IA_NA from 000100011e75e289086266c6f0e6 on br-lan: ok fdb7:90c5:22a1::927/128 
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /etc/hosts - 4 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/dhcp - 2 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/odhcpd - 1 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq-dhcp[1425]: read /etc/ethers - 0 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /etc/hosts - 4 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/dhcp - 2 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq[1425]: read /tmp/hosts/odhcpd - 1 addresses
Fri Apr  8 13:38:39 2016 daemon.info dnsmasq-dhcp[1425]: read /etc/ethers - 0 addresses
Fri Apr  8 13:40:18 2016 daemon.info pppd[1317]: System time change detected.

cokebar, 2016-04-08:
First of all, you should provide your DNS configuration.

From the log you provided, I could tell that there is a problem with your udp ss-tunnel, which can cause DNS resolve problem(when you use it as a DNS relay). This problem may be caused by:
1. UDP features on ss-server was not turned on. Try run ss-server with "-u" parameter.
2. Your ISP block UDP traffic on specific ports/IPs.
3. There's some configuration problem on you router, making udp traffic to remote server unreachable.

oe77, 2016-04-09:
@cokebar 我的配置如下，出现我这种问题应该就如你说的路由器的一些配置问题，使UDP流量远程服务器无法访问，或者可以与OpenWrt Designated Driver 兼容不太好！！

config global
	option global_server 'cfg044a8f'
	option udp_relay_server 'cfg044a8f'

config servers
	option auth_enable '0'
	option server_port 'xxxx'
	option local_port '1081'
	option timeout '600'
	option encrypt_method 'aes-256-cfb'
	option server 'xxxxxxxx'
	option password 'xxxx'

config udp_forward
	option tunnel_port '1153'
	option tunnel_forward '8.8.8.8:53'
	option tunnel_enable '1'

config access_control
	option lan_ac_mode '0'
	option wan_bp_list '/etc/chinadns_chnroute.txt'

config chinadns
	option bidirectional '0'
	option chnroute '/etc/chinadns_chnroute.txt'
	option port '1053'
	option enable '1'
	option server '115.159.157.26,127.0.0.1:1153'

cokebar, 2016-04-09:
@oe77 
1. Try disable UDP relay
2. Try using pdnsd(TCP mode), refer to https://cokebar.info/archives/1053

aa65535, 2016-04-09:
并不支持 ShadowsocksR

closed 2016-04-09 by aa65535