shadowsocks/openwrt-shadowsocks #256: Getting random periodic DNS failures

droseacc, 2016-05-23:
Using OpenWrt Chaos Calmer 15.05.1

I'm getting random DNS failures for long periods without reason, I thought it might be related to ChinaDNS so I removed that and just use Shadowsock's forwarding directly, DNSmasq's DNS forwardings is set to 127.0.0.1#(UDP Local Port) and to not use the resolve file.

This works fine usually but I'm getting random periods of anywhere from 5-30 minutes where the DNS won't work, the dekstop/android client still work while this is happening so doesn't appear to be a server issue. If I wait for a while this usually resolves itself without intervention, also changing server or restarting the router resolves this but it's not an ideal situation obviously. I've checked the logs and nothing comes up related to this.

cokebar, 2016-06-13:
It seems that your ISP or GFW do something with the UDP connections. You can try forward DNS request using TCP instead of UDP. You may need some application like pdnsd/unbound to force using TCP.

Client→→→→→dnsmasq→→→→pdnsd/unbound→→→→→→→→→→→→→ss-server→→remote
             UDP                        UDP                                 TCP(through ss-tunnel)

droseacc, 2016-06-21:
Okay thanks for the reply, I'll give that a try. But why does it start working again if I reset the router, or change server the server in shadowsocks-libev, or why are the desktop and android clients still okay even with UDP?

cokebar, 2016-06-21:
@droseacc , Here's my understanding (may have some misunderstanding):
For desktop and android clients(VPN mode), it uses ss-local. DNS query is on server side. No DNS query is required on client side. So there's no DNS problem for this case.
For ss-spec on router, it uses ss-redir. Some ISP and GFW seem doing some weired things on UDP traffic, making ss UDP traffic unstable. It is my suggestion that try to avoid using UDP over ss unless you really need it.

mengskysama, 2016-07-27:
@droseacc @cokebar 

When I use ss-tunnel to solve domain on remote server I found that UDP pack loss rate is different in different port at the same time. some time even more than 90 percent loss rate in my case. 

the best solution like
Client→(UDP)→dnsmasq-full→(UDP)→pdnsd/unbound→(TCP through ss-redir)→dns spec in pdnsd.
you can get pdnsd bin here https://github.com/mengskysama/openwrt-pdnsd/releases
you can setup step by step with http://blog.mengsky.net/dnsquery-over-tcp/

Luci config should separate DNS setting from Tunnel.

closed 2016-08-14 by aa65535