shadowsocks/openwrt-shadowsocks #258: shadowsocks-libev-spec 2.4.6远程通过OpenVPN连接到路由器后无法通过SS透明代理

dobetter, 2016-05-26:
安装shadowsocks-libev-spec 2.4.6后发现远程通过OpenVPN连接到路由器后无法通过SS透明代理，请问能否修正或者有什么解决办法吗？

cokebar, 2016-05-26:
I think you can directly run ss-server, or other proxy on you router, It's a lot more simple.
VPN software use tun/tap, which creates new interface on your router. By default, ss-spec only proxy LAN traffic an router itself. Traffics on VPN interface are not redirected to ss. You need some iptables rules to make it work properly.

dobetter, 2016-05-26:
Thanks for your help!
Still learning how to add  iptables rules.
Anyone know how to ?

closed 2016-06-04 by aa65535

CzBiX, 2016-07-05:
@dobetter 创建一个包含 `tun0` 的 `interface` 叫 `vpn`，然后在自定义 iptables 里写 `iptables -t nat -A prerouting_vpn_rule -p tcp -j SS_SPEC_WAN_AC`