shadowsocks/openwrt-shadowsocks #263: 运行一段时间后，反应迟缓甚至不能使用的问题

pexcn, 2016-06-04:
在我的ISP，对VPS的`ping`值在`155 ms`左右，使用shadowsocks可以满速，但是运行一段时间之后 (大概一日一夜)，shadowsocks变得反应迟缓，下载速度变得只有几十`KB/s`，再使用一段时间，shadowsocks无法使用了。然后我试了Windows版本的shadowsocks，速度正常可以满速。
通过`/etc/init.d/shadowsocks restart`重启了shadowsocks之后，发现可以使用了，而且速度正常。

aa65535, 2016-06-05:
提供的信息太少，不足以定位问题，或许可以使用 crontab 定时重启。

pexcn, 2016-06-05:
有时间再尝试加`-v`看看log吧...

simonsmh, 2016-06-05:
Sun Jun  5 17:29:34 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:36 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:37 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:39 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:40 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:41 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:43 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:44 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:46 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:29:48 2016 kern.err /usr/bin/ss-redir[15382]: getpeername: Transport endpoint is not connected
Sun Jun  5 17:30:04 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:04 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:12 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:16 2016 kern.err /usr/bin/ss-redir[15382]: server recv: Connection reset by peer
Sun Jun  5 17:30:19 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:29 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:35 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:35 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:38 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:45 2016 kern.err /usr/bin/ss-redir[15382]: remote recv: Connection reset by peer
Sun Jun  5 17:30:47 2016 kern.err /usr/bin/ss-redir[15382]: server recv: Connection reset by peer
Sun Jun  5 17:30:47 2016 kern.err /usr/bin/ss-redir[15382]: server recv: Connection reset by peer
同样的问题。openwrt 150501 ar71xx(wndr4300)。似乎是libev的问题，我在ss-server上也有过类似情况。

pexcn, 2016-06-05:
刚刚想试试 `-v` 抓 log , 很惭愧...原来已经有了，哈哈

pexcn, 2016-06-06:
很惭愧。。。还遇到了个问题，不另开issue了，引用一下：[https://github.com/shadowsocks/shadowsocks-libev/issues/667](https://github.com/shadowsocks/shadowsocks-libev/issues/667)

aa65535, 2016-06-10:
看起来像是 ISP 的问题

pexcn, 2016-06-10:
ISP 的确会达到一定的流量限速，但是我重启 shadowsocks 可以解决，不像是运营商问题...

heke2929, 2016-06-18:
我也遇到这样的问题。

kkjoy, 2016-06-19:
就是运营商，半夜定时重启一下路由器就行了

weijiaming, 2016-07-13:
我也遇到了同样的问题，但我更惨，得20分钟重启一次路由器。

simonsmh, 2016-07-13:
只需要重启ss就行

wongsyrone, 2016-07-20:
看这么多人都有这个问题，我提一下我的思路

对OpenWrt/LEDE来说，重要的库使用自己软件库里面的，比如事件驱动libev和加密库libsodium，其他的libudns和libipset不容易出问题，所以使用ss-libev自带的。libcork作为一个支持，用了上游最新的0.15.0稳定版，以补丁的形式存在。

目前我自己用mbedtls的版本，没任何问题，仅供有兴趣的参考。
详见：https://github.com/wongsyrone/lede-1/tree/master/package/external/shadowsocks-libev/patches 和 对应的Makefile

zxcwdta, 2016-08-01:
表示同样也有这个问题 重启SS, 重启路由, 重启光猫都没法解决.同是本地的一个朋友 我们共享同一个SS 他跑的速度有4000-5000kbps 我只有500kbps

pexcn, 2016-08-01:
升级到最新版之后，好像没这个问题了

simonsmh, 2016-08-01:
请close this issue, thanks.

closed 2016-08-01 by pexcn