shadowsocks/openwrt-shadowsocks #264: Using Guest wireless network

jchia, 2016-06-07:
I configured a Guest wireless network separate from my regular network as described here:
http://blog.ltns.info/linux/guest_ssid_over_openwrt_router/

I find that on my regular network, I can use shadowsocks and visit unharmonious websites like google.com but on the Guest network, I can't. How do I configure my router to have a Guest network that also uses the shadowsocks?

cokebar, 2016-06-08:
I think this is beacause ss-rules only get para "IFNAME" from interface "LAN". see:
https://github.com/shadowsocks/openwrt-shadowsocks/blob/master/files/shadowsocks.rule#L93

@aa65535 In this case, I think there may be an option to let user choose which interface(s) will be proxied.

aa65535, 2016-06-10:
将 `iptables-save -t nat` 的打印结果发上来。

jchia, 2016-06-15:
@cokebar The patch seems to work. I can get to https://google.com/ from the guest network. I see that the patch is essentially repeating the IFNAME stuff replacing lan with guest, however, the part about `-I OUTPUT` is unclear to me. In my ss-rules, the `-I OUTPUT` line is `$IPT -I OUTPUT 1 ${IFNAME:+-o $IFNAME} -p ...` but yours just drops the `${IFNAME:+-o $IFNAME}` instead of repeating the line for `IFNAME2`. I barely understand iptable rules, but does it make more sense to drop the `${IFNAME:+-o $IFNAME}` or repeat the line for `IFNAME2`?

@aa65535 The output has `-A PREROUTING -i br-lan -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC` but no `PREROUTING` rule for `wlan1`. (`lan` is on `br-lan` and `guest` is on `wlan1`.) After applying the patch, I get a new, similar, `PREROUTING` rule for `wlan1`.

jchia, 2016-06-15:
@aa65535 This is the iptables-save output with some parts cut that I think are irrelevant:
`
root@wormhole:/# iptables-save -t nat
# Generated by iptables-save v1.4.21 on Wed Jun 15 10:16:07 2016
*nat
:PREROUTING ACCEPT [101214:7325052]
:INPUT ACCEPT [46712:3406790]
:OUTPUT ACCEPT [38166:2509123]
:POSTROUTING ACCEPT [48592:3039958]
:MINIUPNPD - [0:0]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_guest_rule - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_guest_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_guest_postrouting - [0:0]
:zone_guest_prerouting - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -i br-lan -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC
-A PREROUTING -j delegate_prerouting
-A OUTPUT -o br-lan -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
(omitted rules for -A MINIUPNPD)
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_ac dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 1090
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-lan -j zone_lan_postrouting
-A delegate_postrouting -o pppoe-wan -j zone_wan_postrouting
-A delegate_postrouting -o eth0 -j zone_wan_postrouting
-A delegate_postrouting -o wlan1 -j zone_guest_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-lan -j zone_lan_prerouting
-A delegate_prerouting -i pppoe-wan -j zone_wan_prerouting
-A delegate_prerouting -i eth0 -j zone_wan_prerouting
-A delegate_prerouting -i wlan1 -j zone_guest_prerouting
-A zone_guest_postrouting -m comment --comment "user chain for postrouting" -j postrouting_guest_rule
-A zone_guest_prerouting -m comment --comment "user chain for prerouting" -j prerouting_guest_rule
-A zone_lan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_lan_rule
(omitted rules for -A zone_lan_postrouting)
-A zone_lan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_lan_rule
(omitted rules for -A zone_lan_prerouting)
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -j MINIUPNPD
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
(omitted rules for -A zone_wan_prerouting)
COMMIT
# Completed on Wed Jun 15 10:16:07 2016
`

GuBo88, 2016-06-28:
I would like to use it when connect remote (VPN) to my network that the traffic is than routed thru shadowsocks. I have an VIP account with preferential international connection. Which is must faster tha going out via mobile network directly.
Mobile <-> Home network <-> Shadowsocks
Thanks

commit cebb0c70 (2016-07-27): enhance lan access control

implement LAN muti-policy control && close #188
add -I argument && close #264
closed 2016-07-27 by aa65535