shadowsocks/openwrt-shadowsocks #268: spec版去掉的ss-local到底有什么用途

hdstar2009, 2016-06-27:
原版有个ss-local说是提供了socks代理，那么spec版没有了这个东西，会带来什么影响？
透明代理不能替代socks代理的应用场景？

unitone, 2016-07-05:
spec版去掉ss-local，是全局代理。

hdstar2009, 2016-07-07:
但是现在的不也有全局代理吗 还可以根据chinadns路由表文件不走全局

Randy000, 2016-07-11:
你那只是在路由器运行了ss-local 代替了PC的客户端而已 我这里是要在路由器上做流量转发 用的是ss-redir 不是网关怎么做转发？ 这里要达成接入路由器的设备无需任何额外配置自动翻墙， 如果你是无线AP模式，网关是上层路由，那OpenWRT系统和你PC机就只是同一网段下的不同设备而已，可以等效成OpenWRT和你的PC都是直接插在上层路由设备上的，那么把OpenWRT也看成一台PC，就相当于两台PC接入到上层路由，你说的方式也就相当于一台PC开了shadowsocks客户端然后把端口分享给了局域网里的另一台 那么这里的“另一台”就仍需配置代理 而如果这“另一台”是个PS3 xbox怎么办？哪来的socks5代理客户端让你用？这些设备往往只支持http(s)代理。我这种方案就是要完全抛弃设备上的客户端和任何额外的代理配置

话说你这是多麻烦。这个方案是ss-redir，流量在路由器中做转发，是透明代理；本身并不带ss-local，你在chrome里配置socks5代理当然不能连了

出自https://cokebar.info

fishtorres, 2016-07-13:
我正好就是楼上那种应用场景，所以应该装原版吗

aa65535, 2016-07-13:
ss-local 是用来提供 [SOCKS 代理](https://www.google.com/search?q=SOCKS+%E4%BB%A3%E7%90%86&oq=SOCKS+%E4%BB%A3%E7%90%86&aqs=chrome..69i57&sourceid=chrome&ie=UTF-8)的.

closed 2016-07-13 by aa65535