shadowsocks/openwrt-shadowsocks #269: ss-redir不能正常工作

qianguozheng, 2016-06-27:
我根据这里面的 https://hong.im/ 的配置，设置了ipset, 并且配置了dnsmasq, 使其生效，然后重定向到ss-redir监听的端口。

但是，我发现并ss-redir并没有打印收到任何请求的日志。所以， 我想知道如何调试ss-redir ?
只有如下日志：

```
2016-06-27 17:41:07 INFO: initializing ciphers... aes-256-cfb
 2016-06-27 17:41:07 INFO: tcp port reuse enabled
 2016-06-27 17:41:07 INFO: listening at 0.0.0.0:10800
```
主题思想是：仅将需要重定向到国外的请求，重定向到国外。国内的请求走默认的路由。

配置命令：
ipset -N setmefree iphash
iptables -t nat -A PREROUTING -p tcp -m set --match-set setmefree dst -j REDIRECT --to-port 1080 

shadowsocks.json:
```
{
    "server": "xx.xx.xxx",
    "server_port": 443,
    "local_port": 1080,
    "password": "-------",
    "timeout": 60,
    "method": "aes-256-cfb"
}

```
添加conf-dir=/etc/dnsmasq.d
建立/etc/dnsmasq.d目录并添加测试文件test.conf
```
server=/.google.com/208.67.222.222#443
server=/.google.com.hk/208.67.222.222#443
server=/.gstatic.com/208.67.222.222#443
server=/.google.com/208.67.222.222#443
server=/.google.com.hk/208.67.222.222#443
server=/.gstatic.com/208.67.222.222#443
server=/.ggpht.com/208.67.222.222#443
server=/.googleusercontent.com/208.67.222.222#443
server=/.appspot.com/208.67.222.222#443
server=/.googlecode.com/208.67.222.222#443
server=/.googleapis.com/208.67.222.222#443
server=/.gmail.com/208.67.222.222#443
server=/.google-analytics.com/208.67.222.222#443
server=/.youtube.com/208.67.222.222#443
server=/.googlevideo.com/208.67.222.222#443
server=/.youtube-nocookie.com/208.67.222.222#443
server=/.ytimg.com/208.67.222.222#443
server=/.blogspot.com/208.67.222.222#443
server=/.blogger.com/208.67.222.222#443
server=/.google.co.jp/208.67.222.222#443
server=/.google.co.uk/208.67.222.222#443

ipset=/.google.com/setmefree
ipset=/.google.com.hk/setmefree
ipset=/.gstatic.com/setmefree
ipset=/.ggpht.com/setmefree
ipset=/.googleusercontent.com/setmefree
ipset=/.appspot.com/setmefree
ipset=/.googlecode.com/setmefree
ipset=/.googleapis.com/setmefree
ipset=/.gmail.com/setmefree
ipset=/.google-analytics.com/setmefree
ipset=/.youtube.com/setmefree
ipset=/.googlevideo.com/setmefree
ipset=/.youtube-nocookie.com/setmefree
ipset=/.ytimg.com/setmefree
ipset=/.blogspot.com/setmefree
ipset=/.blogger.com/setmefree
ipset=/.google.co.jp/setmefree
ipset=/.google.co.uk/setmefree

```

经过以上配置后，ss-redir并没有将请求重定向到外网的服务器。

CzBiX, 2016-07-05:
可参考此文配置： https://cokebar.info/archives/664

aa65535, 2016-07-13:
配置文件中添加 `"local_address": "0.0.0.0"`

closed 2016-07-13 by aa65535