shadowsocks/openwrt-shadowsocks #274: 源码资源连接失效

leon0516, 2016-07-18:
第一次编译遇到如下问题
```
failed: Connection timed out.
Retrying.

--2016-07-18 13:17:20--  (try: 4)  https://github-cloud.s3.amazonaws.com/releases/21731829/7b72bc62-29d2-11e6-9d95-ad9ab541de94.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160718%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160718T051614Z&X-Amz-Expires=300&X-Amz-Signature=4880c238c45405aa4935490467ed128087f5a45aeb4146d340ead8892d911468&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-2.4.7.tar.gz&response-content-type=application%2Foctet-stream
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.16.153|:443... failed: Connection timed out.
Retrying.

--2016-07-18 13:17:44--  (try: 5)  https://github-cloud.s3.amazonaws.com/releases/21731829/7b72bc62-29d2-11e6-9d95-ad9ab541de94.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160718%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160718T051614Z&X-Amz-Expires=300&X-Amz-Signature=4880c238c45405aa4935490467ed128087f5a45aeb4146d340ead8892d911468&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-2.4.7.tar.gz&response-content-type=application%2Foctet-stream
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.16.153|:443... failed: Connection timed out.
Giving up.

Download failed.
--2016-07-18 13:18:04--  http://mirror2.openwrt.org/sources/shadowsocks-libev-2.4.7.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-18 13:18:04 ERROR 404: Not Found.

Download failed.
--2016-07-18 13:18:04--  http://downloads.openwrt.org/sources/shadowsocks-libev-2.4.7.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-18 13:18:10 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/leon/workspaces/openwrt-15.05/dl/shadowsocks-libev-2.4.7.tar.gz] Error 2
make[3]: Leaving directory `/home/leon/workspaces/openwrt-15.05/package/shadowsocks-libev'
make[2]: *** [package/shadowsocks-libev/compile] Error 2
make[2]: Leaving directory `/home/leon/workspaces/openwrt-15.05'
make[1]: *** [/home/leon/workspaces/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/leon/workspaces/openwrt-15.05'
[31mBuild failed - please re-run with -j1 to see the real error message[m
make: *** [world] Error 1
```
查看makefile source url并与github上资源对比发现makefile中的连接与github上的不符合，于是将makefile中
`PKG_SOURCE_URL:=https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/v$(PKG_VERSION)`
修改为
`PKG_SOURCE_URL:=https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/$(PKG_VERSION)/$(PKG_SOURCE)`

再次编译又报如下错误
```
78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-18 13:31:38 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/leon/workspaces/openwrt-15.05/dl/shadowsocks-libev-2.4.7.tar.gz] Error 2
make[3]: Leaving directory `/home/leon/workspaces/openwrt-15.05/package/shadowsocks-libev'
make[2]: *** [package/shadowsocks-libev/compile] Error 2
make[2]: Leaving directory `/home/leon/workspaces/openwrt-15.05'
make[1]: *** [/home/leon/workspaces/openwrt-15.05/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/leon/workspaces/openwrt-15.05'
[31mBuild failed - please re-run with -j1 to see the real error message[m
make: *** [world] Error 1
```
我不太清除下载的机制如何，个人猜测，是在github上下载不了，然后访问downloads.openwrt.org下载源码，但我打开downloads.openwrt.org查看了文件，并没有发现shadowsocks-libev-2.4.7.tar.gz文件。
最终的解决办法是在dl文件夹手动下载shadowsocks-libev-2.4.7.tar.gz源码文件，然后编译就通过了。

aa65535, 2016-07-18:
无法复现，可以试试 `wget https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/v2.4.7/shadowsocks-libev-2.4.7.tar.gz` 是否可以下载。
```
make[1]: Entering directory '/home/rin/openwrt/wndr4300/sdk'
make[2]: Entering directory '/home/rin/openwrt/wndr4300/sdk/package/shadowsocks-libev'
mkdir -p /home/rin/openwrt/wndr4300/sdk/dl
/home/rin/openwrt/wndr4300/sdk/scripts/download.pl "/home/rin/openwrt/wndr4300/sdk/dl" "shadowsocks-libev-2.4.7.tar.gz" "e08ba321cb8b7778597942d529cedfba" "https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/v2.4.7"
--2016-07-18 16:27:52--  https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/v2.4.7/shadowsocks-libev-2.4.7.tar.gz
Resolving github.com (github.com)... 192.30.253.112
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/21731829/7b72bc62-29d2-11e6-9d95-ad9ab541de94.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160718%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160718T082802Z&X-Amz-Expires=300&X-Amz-Signature=4250667933b071702c231db8cd76f396c8c4af2c44a14c5b2be6a1c45a1bf6d0&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-2.4.7.tar.gz&response-content-type=application%2Foctet-stream [following]
--2016-07-18 16:27:55--  https://github-cloud.s3.amazonaws.com/releases/21731829/7b72bc62-29d2-11e6-9d95-ad9ab541de94.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160718%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160718T082802Z&X-Amz-Expires=300&X-Amz-Signature=4250667933b071702c231db8cd76f396c8c4af2c44a14c5b2be6a1c45a1bf6d0&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dshadowsocks-libev-2.4.7.tar.gz&response-content-type=application%2Foctet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.50.58
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.50.58|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 820529 (801K) [application/octet-stream]
Saving to: 'STDOUT'

     0K .......... .......... .......... .......... ..........  6% 37.7K 20s
    50K .......... .......... .......... .......... .......... 12% 56.7K 15s
   100K .......... .......... .......... .......... .......... 18% 40.5K 15s
   150K .......... .......... .......... .......... .......... 24% 71.3K 12s
   200K .......... .......... .......... .......... .......... 31% 35.7K 12s
   250K .......... .......... .......... .......... .......... 37% 54.3K 11s
   300K .......... .......... .......... .......... .......... 43%  145K 9s
   350K .......... .......... .......... .......... .......... 49% 71.7K 8s
   400K .......... .......... .......... .......... .......... 56% 61.6K 6s
   450K .......... .......... .......... .......... .......... 62%  207K 5s
   500K .......... .......... .......... .......... .......... 68%  121K 4s
   550K .......... .......... .......... .......... .......... 74% 45.0K 3s
   600K .......... .......... .......... .......... .......... 81% 6.30M 2s
   650K .......... .......... .......... .......... .......... 87%  200K 1s
   700K .......... .......... .......... .......... .......... 93% 5.85M 1s
   750K .......... .......... .......... .......... .......... 99% 64.1K 0s
   800K .                                                     100% 2475G=11s

2016-07-18 16:28:09 (72.0 KB/s) - written to stdout [820529/820529]
```

closed 2016-07-18 by aa65535

leon0516, 2016-07-19:
抱歉啊，耽误你时间了，以后出现问题我会多试几次。