shadowsocks/openwrt-shadowsocks #280: 更新到2.4.8之后 编译提示no rule to make traget

zxcwdta, 2016-08-01:
2.4.7正常 但是2.4.8和2.4.8-2都提示这个错误

zxcwdta, 2016-08-01:
make[3]: Entering directory `/home/dta/openwrt/custom/openwrt-shadowsocks-2.4.8-2'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float  -I/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float  -I/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/dta/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib " make   -C /home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/. AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" ;
make[4]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8'
make  all-recursive
make[5]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8'
Making all in libsodium
make[6]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium'
Making all in src
make[7]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src'
Making all in libsodium
make[8]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium'
Making all in include
make[9]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium/include'
make[9]: Nothing to be done for `all'.
make[9]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium/include'
make[9]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium'
make[9]: Nothing to be done for `all-am'.
make[9]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium'
make[8]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src/libsodium'
make[8]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src'
make[8]: Nothing to be done for `all-am'.
make[8]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src'
make[7]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium/src'
make[7]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium'
make[7]: Nothing to be done for `all-am'.
make[7]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium'
make[6]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libsodium'
Making all in libcork
make[6]: Entering directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libcork'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/libcork'
Making all in libipset
[... cut ...]
make[3]: *** [/home/dta/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.8/.built] Error 2
make[3]: Leaving directory `/home/dta/openwrt/custom/openwrt-shadowsocks-2.4.8-2'
make[2]: *** [package/feeds/custom/openwrt-shadowsocks-2.4.8-2/compile] Error 2
make[2]: Leaving directory `/home/dta/openwrt'
make[1]: *** [/home/dta/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/dta/openwrt'
make: *** [world] Error 2

aa65535, 2016-08-01:
大概是系统缺少 asciidoc 包。
2.4.8 多了个 doc 目录。

zxcwdta, 2016-08-02:
谢谢解答 安装了asciidoc之后编译顺利通过

commit 8f08522f (2016-08-09): disable documentation build, fix #280
closed 2016-08-02 by zxcwdta