shadowsocks/openwrt-shadowsocks #281: 2.4.8-2版实际使用后发现重启后失效

kghx, 2016-08-01:
在OP的BB版和CC版上均测试过，都是这个问题，
已设置为开机自启动，配置完参数后能正常运行，
但是一旦重启路由，查看系统进程发现已经自动运行，但是实际已无效，
再在luci界面点击一次“保存&应用“后，立即又恢复正常了。

该问题已多次测试，均存在。

aa65535, 2016-08-01:
提供一下 `iptables-save -t nat` 的打印结果。

kghx, 2016-08-02:
重启路由前一切正常，重启路由后虽然相关进程已经启动，但是已失效，
下面是 iptables-save -t nat  打印结果
/root$ iptables-save -t nat
# Generated by iptables-save v1.4.21 on Tue Aug  2 13:56:19 2016
*nat
:PREROUTING ACCEPT [440462:85471656]
:INPUT ACCEPT [9641:823202]
:OUTPUT ACCEPT [1775:136253]
:POSTROUTING ACCEPT [1304:102844]
:MINIUPNPD - [0:0]
:SS_SPEC_LAN_AC - [0:0]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -j delegate_prerouting
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_bp src -j RETURN
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_fw src -j SS_SPEC_WAN_FW
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_no src -j SS_SPEC_WAN_AC
-A SS_SPEC_LAN_AC -j SS_SPEC_WAN_AC
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_fw dst -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_bp dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -m set --match-set ss_spec_wan_sp dst -j RETURN
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 1080
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-lan -j zone_lan_postrouting
-A delegate_postrouting -o eth0 -j zone_wan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-lan -j zone_lan_prerouting
-A delegate_prerouting -i eth0 -j zone_wan_prerouting
-A zone_lan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_lan_rule
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -j MINIUPNPD
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
COMMIT
# Completed on Tue Aug  2 13:56:19 2016

下面是相关系统进程状态：

2446  
root  
/usr/bin/ss-redir -c /var/etc/shadowsocks.json -f /var/run/ss-redir.pid  
0%  
5%  
 
2461  
root  
/usr/bin/ss-tunnel -c /var/etc/shadowsocks.json -u -l 5354 -L 8.8.8.8:53 -f /var/run/ss-tunnel.pid  
0%  
5% 

2259  
root  
/usr/bin/chinadns -p 5353 -s 114.114.114.114,127.0.0.1#5354 -c /etc/chinadns_chnroute.txt -m  
0%  
1%

aa65535, 2016-08-02:
@kghx 分别执行下面两个命令，看看有没有正常输出

`uci get -P/var/state network.lan.ifname`

`uci -P/var/state get network.lan.ifname`

kghx, 2016-08-02:
均正常输出了lan接口

/root$  uci get -P/var/state network.lan.ifname 
br-lan
/root$ uci -P/var/state get network.lan.ifname
br-lan

aa65535, 2016-08-02:
应该是 /var/state/network 在 ss 启动的时候还没准备好，上面的目录获取不到 br-lan

kghx, 2016-08-02:
尝试将启动优先级改为99，测试还是不行。

aa65535, 2016-08-02:
改优先级是没用的，那个时间没差多少。

commit 0ddf32c7 (2016-08-04): fix #281
closed 2016-08-04 by aa65535