shadowsocks/openwrt-shadowsocks #285: 127.0.0.1:5300无响应

nsecgo, 2016-08-06:
[image]
配置如上图，ss服务器是正常的。然后
nsec@nsecgo:~$ dig @192.168.1.1 -p 5300 google.com

; <<>> DiG 9.10.3-P4-Ubuntu <<>> @192.168.1.1 -p 5300 google.com
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached

请问应该怎么解决这个问题

pexcn, 2016-08-06:
估计是你ISP限制了UDP吧？

nsecgo, 2016-08-06:
@pexcn 我又把udp服务器换成我国内的ss服务器，udp转发到国内的dns，然后
dig @127.0.0.1 -p 5300 www.baidu.com 返回正常。
所以这有没有可能是我国外ss服务器的问题，还是就是你说的isp限制？

pexcn, 2016-08-06:
ISP限制国外服务器的UDP流量，我这里的ISP就是这样

nsecgo, 2016-08-06:
@pexcn 怎么验证isp限制了国外udp流量？
我在国外vps上用nc监听了一个udp端口，然后在本地是可以连上的

pexcn, 2016-08-06:
`dig google.com @8.8.4.4 +notcp` 如果超时，就限制了

nsecgo, 2016-08-06:
@pexcn 正常返回了，然后我又把路由器上的ss关了，还是返回正常，两次返回的ip都是正确的google ip

pexcn, 2016-08-06:
我这的ISP是达到一定流量才会限制，不是总是限制

closed 2016-08-07 by nsecgo