shadowsocks/openwrt-shadowsocks #288: 2.4.8-2版使用后发现重启后失效且无法恢复

liyiecho, 2016-08-08:
默认是开机自启动，之前配置完参数后可以正常运行，但是第二天使用时，发现失效了。怎么重启SS跟路由都无法正常使用，甚至重装都没用。

下面是 iptables-save -t nat 打印结果：
root@OpenWrt:~# iptables-save -t nat
# Generated by iptables-save v1.4.21 on Mon Aug  8 16:07:45 2016
*nat
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [25:1757]
:POSTROUTING ACCEPT [25:1757]
:SS_SPEC_LAN_AC - [0:0]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -i br-lan -p tcp -j SS_SPEC_LAN_AC
-A PREROUTING -j delegate_prerouting
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_bp src -j RETURN
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_fw src -j SS_SPEC_WAN_FW
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_no src -j SS_SPEC_WAN_AC
-A SS_SPEC_LAN_AC -j SS_SPEC_WAN_AC
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_fw dst -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_bp dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -m set --match-set ss_spec_wan_sp dst -j RETURN
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 1080
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-lan -j zone_lan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-lan -j zone_lan_prerouting
-A zone_lan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_lan_rule
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
COMMIT
# Completed on Mon Aug  8 16:07:45 2016

下面是相关系统进程状态：
627	root	/usr/bin/chinadns -p 5353 -s 114.114.114.114,127.0.0.1:5300 -c /etc/chinadns_chnroute.txt -d -m	0%	0%
754	root	/usr/bin/ss-redir -c /var/etc/shadowsocks.json -f /var/run/ss-redir.pid	0%	1%
765	root	/usr/bin/ss-tunnel -c /var/etc/shadowsocks.json -u -l 5300 -L 8.8.4.4:53 -f /var/run/ss-tunnel.pid	0%	1%

分别执行下面两个命令：
uci get -P/var/state network.lan.ifname
uci -P/var/state get network.lan.ifname
均正常输出了lan接口
/root$ uci get -P/var/state network.lan.ifname 
br-lan
/root$ uci -P/var/state get network.lan.ifname
br-lan

aa65535, 2016-08-08:
iptables 规则没问题，看一下下面三个命令的输出是什么
```
ipset save ss_spec_lan_bp
ipset save ss_spec_lan_fw
ipset save ss_spec_lan_no
```

另外排查一下服务器是否可以正常连接。

liyiecho, 2016-08-08:
分别是这样的：
```
root@OpenWrt:~#ipset save ss_spec_lan_bp
create ss_spec_lan_bp hash:net family inet hashsize 64 maxelem 65536
root@OpenWrt:~# ipset save ss_spec_lan_fw
create ss_spec_lan_fw hash:net family inet hashsize 64 maxelem 65536
root@OpenWrt:~# ipset save ss_spec_lan_no
create ss_spec_lan_no hash:net family inet hashsize 64 maxelem 65536
```

服务器正常连接的，手机电脑可以通过ss客户端正常使用的

liyiecho, 2016-08-10:
因为接入的设备没有获取到正确的dns，所以导致ss失效

closed 2016-08-10 by liyiecho