shadowsocks/openwrt-shadowsocks #290: 2.4.8-2不能正常使用。

FreeVicto, 2016-08-13:
新版本：
luci-app-shadowsocks-spec	1.5.0-1
shadowsocks-libev-spec-polarssl	2.4.8-2
总是不能正常使用，进程存在，启动用-v 也看不出问题。试了几次都不行。

老版本
luci-app-shadowsocks-spec	1.4.0-1
shadowsocks-libev-spec-polarssl	2.4.7-1
可以正常使用，，

无奈不知道问题在哪里，也不知道如何找，只能--force-downgrade用前一版本；

aa65535, 2016-08-13:
参考 #281

看一下 `iptables-save -t nat` 的打印结果。

FreeVicto, 2016-08-13:
@aa65535 

root@NemoOpenWrt505_2nd:/tmp# iptables-save -t nat
# Generated by iptables-save v1.4.21 on Sat Aug 13 21:03:21 2016
*nat
:PREROUTING ACCEPT [23:1204]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [1:76]
:POSTROUTING ACCEPT [0:0]
:SS_SPEC_LAN_AC - [0:0]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:postrouting_wifi_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:prerouting_wifi_rule - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
:zone_wifi_postrouting - [0:0]
:zone_wifi_prerouting - [0:0]
-A PREROUTING -j delegate_prerouting
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_bp src -j RETURN
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_fw src -j SS_SPEC_WAN_FW
-A SS_SPEC_LAN_AC -m set --match-set ss_spec_lan_no src -j SS_SPEC_WAN_AC
-A SS_SPEC_LAN_AC -j SS_SPEC_WAN_AC
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_fw dst -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_bp dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -m set --match-set ss_spec_wan_sp dst -j RETURN
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 1080
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-wifi -j zone_wifi_postrouting
-A delegate_postrouting -o eth1 -j zone_wan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-wifi -j zone_wifi_prerouting
-A delegate_prerouting -i eth1 -j zone_wan_prerouting
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
-A zone_wifi_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wifi_rule
-A zone_wifi_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wifi_rule
COMMIT
# Completed on Sat Aug 13 21:03:21 2016

root@NemoOpenWrt505_2nd:/tmp# uci get -P/var/state network.lan.ifname
uci: Entry not found
root@NemoOpenWrt505_2nd:/tmp# uci -P/var/state get network.lan.ifname
uci: Entry not found

似乎是SS启动时候，网络还没有准备好，应该如何处理？

FreeVicto, 2016-08-13:
补充说明：
我在旧版本的另一台正常使用的路由器上得到如下结果：

root@NemoOpenWrt505:~# iptables-save -t nat
# Generated by iptables-save v1.4.21 on Sat Aug 13 21:12:31 2016
*nat
:PREROUTING ACCEPT [3192:448171]
:INPUT ACCEPT [3437:267582]
:OUTPUT ACCEPT [2705:251149]
:POSTROUTING ACCEPT [299:20465]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:postrouting_wifi_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:prerouting_wifi_rule - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
:zone_wifi_postrouting - [0:0]
:zone_wifi_prerouting - [0:0]
-A PREROUTING -p tcp -j SS_SPEC_WAN_AC
-A PREROUTING -s 192.168.2.0/24 -p udp -m udp --dport 53 -j DNAT --to-destination 192.168.2.1
-A PREROUTING -s 192.168.2.0/24 -p tcp -m tcp --dport 53 -j DNAT --to-destination 192.168.2.1
-A PREROUTING -j delegate_prerouting
-A OUTPUT -p tcp -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_ac dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 1080
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-wifi -j zone_wifi_postrouting
-A delegate_postrouting -o eth1 -j zone_wan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-wifi -j zone_wifi_prerouting
-A delegate_prerouting -i eth1 -j zone_wan_prerouting
-A zone_wan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wan_rule
-A zone_wan_postrouting -j MASQUERADE
-A zone_wan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wan_rule
-A zone_wifi_postrouting -m comment --comment "user chain for postrouting" -j postrouting_wifi_rule
-A zone_wifi_prerouting -m comment --comment "user chain for prerouting" -j prerouting_wifi_rule
COMMIT
# Completed on Sat Aug 13 21:12:31 2016
root@NemoOpenWrt505:~# uci get -P/var/state network.lan.ifname
uci: Entry not found
root@NemoOpenWrt505:~# uci -P/var/state get network.lan.ifname
uci: Entry not found

视乎与网络是否准备好无关，D-LINK的DR505无线路由只有一个网口，eth1和br-wifi.

FreeVicto, 2016-08-13:
uci get -P/var/state network.wifi.ifname,倒是有输出，建一个lan口给ss用吗？

aa65535, 2016-08-13:
这个问题在后面的更新里已经修复。

aa65535, 2016-08-13:
@FreeVicto 麻烦提供一下在 DR505 上以下这些命令的输出结果

 1. `uci show firewall | grep zone | grep name`
 2. `uci show network | grep -v loopback | grep -v globals`
 3. `brctl show`
 4. `cat /var/state/network`

FreeVicto, 2016-08-14:
@aa65535 
root@NemoOpenWrt505_2nd:~# uci show firewall | grep zone | grep name
firewall.@zone[0].name=wifi
firewall.@zone[1].name=wan
root@NemoOpenWrt505_2nd:~# uci show network | grep -v loopback | grep -v globals
network.wan=interface
network.wan.ifname=eth1
network.wan._orig_ifname=eth1
network.wan._orig_bridge=false
network.wan.proto=dhcp
network.wifi=interface
network.wifi.proto=static
network.wifi.netmask=255.255.255.0
network.wifi.ipaddr=192.168.3.1
network.wifi.gateway=192.168.3.1
network.wifi.type=bridge
root@NemoOpenWrt505_2nd:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-wifi         7fff.c8d3a365d86d       no              wlan0
root@NemoOpenWrt505_2nd:~# cat /var/state/network
network.loopback.up=1
network.loopback.device=lo
network.wifi.up=1
network.wifi.ifname=br-wifi
network.wan.up=1
network.wan.device=eth1

aa65535, 2016-08-14:
已修复

closed 2016-08-14 by aa65535