shadowsocks/openwrt-shadowsocks #291: 配置完成后似乎该不能上的网站还是上不了呢

caoli5288, 2016-08-14:
浏览器插件里面可以扶墙上网，路由器里面就不行了的说，出口ip已经是服务器的ip了，是dns问题吗

aa65535, 2016-08-14:
根据您的描述我无法猜测原因，如果您认为 DNS 有问题，可以先解决 DNS 的问题后再试。

caoli5288, 2016-08-14:
@aa65535 貌似不是dns问题，我将dns设置成8.8.4.4和8.8.8.8了，还是不可以呢。百度搜索本机ip，看到出口ip已经是位于米国的服务器地址了，但是。。。
```
$ ping www.google.com
PING www.google.com (93.46.8.89) 56(84) bytes of data.
^C
--- www.google.com ping statistics ---
14 packets transmitted, 0 received, 100% packet loss, time 12999ms
```
对路由器翻墙也就这两天才接触的，不是很熟悉，如果可以请帮助我，谢谢。

aa65535, 2016-08-14:
`93.46.8.89` 这个 ip 不是 Google 的，说明 DNS 污染了。

可以参考 https://cokebar.info/archives/664

closed 2016-08-14 by aa65535

caoli5288, 2016-08-14:
我有一个疑问，为什么浏览器插件的socks代理不会遇到这个问题呢

aa65535, 2016-08-14:
@caoli5288 因为「远程解析」

caoli5288, 2016-08-14:
那么路由器可以实现类似的效果吗？ChinaDNS我昨天试了也是一样没办法访问呢

aa65535, 2016-08-14:
https://sourceforge.net/p/openwrt-dist/wiki/Plan5/

同时，您需要将下游设备的 DNS 服务器设置为 路由器IP

caoli5288, 2016-08-14:
@aa65535 dhcp分配到的dns就是路由器的呢

aa65535, 2016-08-14:
网上教程很多，可以去参考一下。

caoli5288, 2016-08-14:
@aa65535 总之是感谢了，我去试试vpn方案看看