shadowsocks/openwrt-shadowsocks #292: 貌似问题越改越大，，，

FreeVicto, 2016-08-16:
如我上一个提出的问题：
/*******************************************************************************/
新版本：
luci-app-shadowsocks-spec 1.5.0-1
shadowsocks-libev-spec-polarssl 2.4.8-2
总是不能正常使用，进程存在，启动用-v 也看不出问题。试了几次都不行。

老版本
luci-app-shadowsocks-spec 1.4.0-1
shadowsocks-libev-spec-polarssl 2.4.7-1
可以正常使用，，
/*******************************************************************************/

今天欣喜看到2.4.8-3的更新，，以为解决问题了，，，结果发现sourceforge上没有libev-spec-polarssl 2.4.8-3的版本，

只好换用shadowsocks-libev-polarssl_2.4.8-3，这样luci-app-shadowsocks-spec 1.5.0-1又提示没有shadowsocks的spec版本，只好改换luci-app-shadowsocks	-1.1.0-1， 发现还是不能正常使用。。

无奈退回到libev-spec-polarssl 2.4.7-1，，，不知道为何不提供libev-spec-polarssl 2.4.8-3的更新了？

pexcn, 2016-08-16:
并没有问题啊

alexwhen, 2016-08-16:
shadowsocks-libev-spec只有2.4.8-2没有2.4.8-3

aa65535, 2016-08-16:
@FreeVicto 在「访问控制」里将需要使用代理的接口选择上.

aa65535, 2016-08-16:
@FreeVicto @alexwhen
ss-spec 已经停止维护，详情参考 https://github.com/shadowsocks/luci-app-shadowsocks/wiki

closed 2016-08-16 by aa65535

FreeVicto, 2016-08-18:
@pexcn 你好,请问你装的是哪个版本还有luci是哪个? 希望得到你的指导..

pexcn, 2016-08-18:
@FreeVicto 不是spec，是那个新版，没有问题。