shadowsocks/openwrt-shadowsocks #309: LEDE sdk compilation error

wtjfree, 2016-11-28:
What version of shadowsocks-libev are you using?

shadowsocks-libev for lede
What operating system are you using?

Ubuntu 16.04.1 LTS x64
What did you do?

ubuntu下安装了:
build-essential libssl-dev libncurses5-dev unzip subversion mercurial gawk zlib1g-dev

按照shadowsocks-libev for openwrt编译操作步骤(sdk有所不同):
tar xjf lede-sdk-ar71xx-generic_gcc-5.4.0_musl-1.1.15.Linux-x86_64.tar.bz2
cd lede-sdk-ar71xx-generic_gcc-5.4.0_musl-1.1.15.Linux-x86_64
./scripts/feeds update packages
./scripts/feeds install libpcre
git clone https://github.com/shadowsocks/openwrt-shadowsocks.git package/shadowsocks-libev
make menuconfig
Network -> shadowsocks-libev
make package/shadowsocks-libev/compile V=99
What did you expect to see?

成功编译shadowsocks-libev-2.5.5 for lede ipk文件
What did you see instead?

checking for strlcat... yes
checking for strlcpy... yes
checking pkg-config is at least version 0.9.0... yes
configure: error: Invalid kernel source directory /lib/modules/4.4.0-21-generic/source
Makefile:53: recipe for target '/home/openwrt/source/build_dir/target-mips_24kc_musl-1.1.15/linux-lantiq_xrx200/ipset-6.30/.configured_yyyyyyy' failed
make[2]: *** [/home/openwrt/source/build_dir/target-mips_24kc_musl-1.1.15/linux-lantiq_xrx200/ipset-6.30/.configured_yyyyyyy] Error 1
make[2]: Leaving directory '/home/openwrt/source/package/network/utils/ipset'
package/Makefile:127: recipe for target 'package/network/utils/ipset/compile' failed
make[1]: *** [package/network/utils/ipset/compile] Error 2
make[1]: Leaving directory '/home/openwrt/source'
/home/openwrt/source/include/toplevel.mk:194: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2
What is your config in detail (with all sensitive info masked)?

n/a

wongsyrone, 2016-11-29:
ipset->kmod-ipt-ipset->kernel

很明显sdk没有内核源码树，去掉ipset就行了

wtjfree, 2016-11-29:
您的意思是不是在make menuconfig里面去掉这个ipset就可以了?

wongsyrone, 2016-11-29:
ipset选的依赖也要去掉，如果想要这个，用源码树自己编译

wtjfree, 2016-11-29:
在make menuconfig里面选择编译任意shadowsocks-libev-server的选项ipset都会被选上而且不可以取消..

wtjfree, 2016-11-29:
问题已经解决感谢热心帮助. 最终还是少各种包.建议作者能帮忙列出依赖包以方便后来者.

closed 2016-11-29 by wtjfree