shadowsocks/openwrt-shadowsocks #310: LEDE precompiled release

cokebar, 2016-12-09:
Any plan for this?

lixingcong, 2016-12-09:
LEDE正式版都没出来 用的人没那么多吧

wongsyrone, 2016-12-22:
LEDE is planning a stable release

http://lists.infradead.org/pipermail/lede-dev/2016-December/004790.html

wtjfree, 2016-12-23:
lede 可以自己用sdk编译的。没啥问题。就是编译环境设置的蛋疼。

aa65535, 2016-12-26:
需要等到稳定版发布。

closed 2016-12-27 by aa65535

wongsyrone, 2017-01-02:
mbedtls的补丁问题正在LEDE讨论，有兴趣的也可以去表达意见

https://github.com/lede-project/source/pull/657

cokebar, 2017-01-02:
@aa65535 http://openwrt-dist.sourceforge.net/packages/LEDE/ 里面的libmbedtls，支持RC4、CFB这些么，能用RC4-MD5么

wongsyrone, 2017-01-02:
一旦正确编译了ss-libev，同时链接的libmbedtls就可以支持原来polarssl的所有加密。aa65535预编译的已经做好了，有兴趣也可以自己编译。

wongsyrone, 2017-01-02:
有一个坑是，如果安装了官方编译的mbedtls而不是自己编译的，会因为默认干掉了几种不安全的功能，导致绝大部分加密不可用。

cokebar, 2017-01-02:
@wongsyrone 这个我知道，但是自编译的会有checksum报错，虽说不是不能装但是还是不爽。
而且我不知道具体哪些是mbedtls（不带lede那个patch的）支持的；哪些又是因为打了那个patch导致不支持了的。

aa65535, 2017-01-04:
@cokebar checksum 报错大概是因为你使用的是 OpenWrt 15.05.1 及以下版本的 SDK, 因为目前 mbedtls 的 Makefile 中 PKG_MD5SUM 的字段改为了 sha256sum 而不是之前的 md5sum, 所以旧版本的 SDK 不支持。

wongsyrone, 2017-01-04:
他说的checksum应该不是你说的问题，PKG_MD5SUM是向后兼容的，应该是依赖里面有iptables的模块，导致依赖特定的内核版本。 无论如何 --force-checksum 都是不推荐的，除非你知道在干什么。

cokebar, 2017-01-04:
@wongsyrone 我不是太懂这个checksum指的是什么对象的hash值，最终生成的ipk的？还是源码包的？ 是我是修改了那个patch导致checksum不对？ 
我用的SDK应该和我刷机镜像一致的4.4.38版本内核啊

wongsyrone, 2017-01-04:
如果有错误提示可以贴出来看看，如果是指Makefile里面的PKG_MD5SUM的话，这个只是在拉取源码的时候做校验的，以防数据不完整或被篡改。patch一般是独立于源码存在的，一般用途是适合交叉编译或者是做一些定制。一般的patch都会发送给上游，只有不适合的才会留在OpenWrt的源码树里面。

debiansid, 2017-01-04:
@wongsyrone  when you will update shadowsocks-libev in your lede-1 to 2.6.0? thanks

wongsyrone, 2017-01-17:
LEDE Update
===

Hi list,

this is just a heads-up to inform you that LEDE master has been branched
into a new branch "lede-17.01" now.

A buildbot system has been set up at
http://release-builds.lede-project.org/17.01/images which will begin
building images and SDKs shortly.

Once the base phase1 builds for each architecture have been finished
within the next two to three days, we will start building the package
repositories for the release branch.

The next 7 days shall be used to work on finalizing the "lede-17.01"
branch and to cherry-pick important outstanding fixes from master.

The plan is to have test builds available until next Monday, the 23th
and to follow up with the final release decision on the same day.

If all goes well, there will be a one week testing period following the
23th until the 30th. If no major issues are found, LEDE v17.01.0 can
then be tagged on Monday, the 30th and final builds shall be available
shortly after.

If serious issues arise during the first testing period, we might decide
to extended by the test phase by a further week until a resolution is found.

If targets or specific images cannot be fixed in time, we should
consider dropping them from the initial release and aim for
adding/fixing them in the next LEDE v17.01.1 point release.

Thanks,
Jo

_______________________________________________
Lede-dev mailing list
Lede-dev@lists.infradead.org
http://lists.infradead.org/mailman/listinfo/lede-dev