shadowsocks/openwrt-shadowsocks #312: build process failed , please help me

annbigbig, 2016-12-23:
這是最後幾行錯誤訊息
可以幫我看一下嗎？
i don't know what's going wrong, please help me

make[3]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6'
Makefile:83: recipe for target '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/.built' failed
make[2]: *** [/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/.built] Error 2
make[2]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/package/shadowsocks-libev'
package/Makefile:196: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/include/toplevel.mk:185: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

aa65535, 2016-12-23:
编译需要系统提前安装有相应的软件包。
对照这里安装即可 https://wiki.openwrt.org/doc/howto/buildroot.exigence

closed 2016-12-23 by aa65535

wongsyrone, 2016-12-23:
> No targets specified and no makefile found

命令可能敲错了

annbigbig, 2016-12-23:
可是我確定我都安裝了呀
會不會是什麼別的原因呢？
```
root@labasky-Lenovo-B590:~# apt-get install git-core build-essential libssl-dev libncurses5-dev unzip gawk
正在讀取套件清單... 完成
正在重建相依關係          
正在讀取狀態資料... 完成
build-essential is already the newest version (12.1ubuntu2).
gawk is already the newest version (1:4.1.3+dfsg-0.1build1).
git-core is already the newest version (1:2.9.3-1).
libncurses5-dev is already the newest version (6.0+20160625-1ubuntu1).
libssl-dev is already the newest version (1.0.2g-1ubuntu9).
unzip is already the newest version (6.0-20ubuntu1).
升級 0 個，新安裝 0 個，移除 0 個，有 66 個未被升級。
```

aa65535, 2016-12-23:
还有下面表格中列出的软件包
```
sudo apt-get install libxml2-dev libcppunit-dev autoconf automake autotools-dev autopoint libtool
```

wongsyrone, 2016-12-23:
最好把你敲make开始的所有输出都贴上

annbigbig, 2016-12-23:
感謝兩位大俠
我人在外面工作
晚一點我把完整的錯誤信息貼上
感謝大家的熱情相助

annbigbig, 2016-12-23:
輸出訊息很多，所以我執行
$ make -j1 V=s | tee build.log

然後這是build.log的內容
```
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64$ cat ./build.log 
#
# configuration written to .config
#
make[1]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
make[2]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/package/shadowsocks-libev'
mkdir -p /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/dl
/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/download.pl "/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/dl" "shadowsocks-libev-2.5.6.tar.gz" "bb99e090640c8af8d7da961a6230c70b" "" "https://github.com/shadowsocks/openwrt-shadowsocks/releases/download/v2.5.6"
. /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/include/shell.sh; gzip -dc /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/dl/shadowsocks-libev-2.5.6.tar.gz | tar -C /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/.. -xf - 
touch /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/.prepared_e2719b1700a36426a77a249b339fe7fc
rm -f /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp/.shadowsocks-libev.openssl_installed
(cd /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autom4te AUTOCONF=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autoconf AUTOMAKE=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/automake ACLOCAL=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/aclocal AUTOHEADER=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/libtoolize LIBTOOL=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/libtool M4=/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/m4 AUTOPOINT=true /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/share/aclocal -I /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/host/share/aclocal -I /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/share/aclocal -I m4 -I . . || true; fi; );
(cd /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/./; if [ -x ./configure ]; then find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/ -name config.guess | xargs -r chmod u+w; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/ -name config.guess | xargs -r -n1 cp --remove-destination /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/config.guess; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/ -name config.sub | xargs -r chmod u+w; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/ -name config.sub | xargs -r -n1 cp --remove-destination /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2 [... cut ...]
make[2]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/package/shadowsocks-libev'
package/Makefile:196: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/include/toplevel.mk:185: recipe for target 'package/shadowsocks-libev/compile' failed
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64$ 
```

希望大家指點一下，非常感謝

annbigbig, 2016-12-23:
您另外說的這些套件我確定有安裝
所以是我作錯了什麼動作嗎？

```
root@labasky-Lenovo-B590:~# apt-get install autoconf automake autotools-dev autopoint libtool
正在讀取套件清單... 完成
正在重建相依關係          
正在讀取狀態資料... 完成
autoconf is already the newest version (2.69-10).
autoconf 被設定為手動安裝。
automake is already the newest version (1:1.15-4ubuntu1).
automake 被設定為手動安裝。
autopoint is already the newest version (0.19.8.1-1ubuntu2).
autopoint 被設定為手動安裝。
autotools-dev is already the newest version (20160430.1).
autotools-dev 被設定為手動安裝。
libtool is already the newest version (2.4.6-1).
升級 0 個，新安裝 0 個，移除 0 個，有 66 個未被升級。
root@labasky-Lenovo-B590:~# 

```

wongsyrone, 2016-12-23:
> bash: /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autoreconf: No such file or directory

SDK缺少文件

annbigbig, 2016-12-24:
我手動作連結過去
`$ ln -s /usr/bin/autoreconf /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/autoreconf`

還有我發現這個aclocal應該也少了
`$ ln -s /usr/bin/aclocal /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/host/bin/aclocal`

然後我回SDK根目錄再試一次
```
$ make clean
$ make package/shadowsocks-libev/compile V=99 | tee build.log
```

但是結果還是失敗，錯誤訊息如下
```
#
# configuration written to .config
#
make[1]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
make[2]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/package/shadowsocks-libev'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6:shadowsocks-libev-2.5.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6:shadowsocks-libev-2.5.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include " LDFLAGS="-L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro " make   -C /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6:shadowsocks-libev-2.5.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6'
make[3]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/shadowsocks-libev/openssl/shadowsocks-libev-2.5.6'
[... cut ...]
make[2]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/package/shadowsocks-libev'
make[1]: *** [package/shadowsocks-libev/compile] Error 2
package/Makefile:196: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/include/toplevel.mk:185: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

```

大俠請問有解嗎？

aa65535, 2016-12-25:
@annbigbig SDK 不完整，重新下载吧。

annbigbig, 2016-12-25:
@aa65535 大俠，我是從OpenWRT官網下載的呢，所以我還能去那裡下載呢？
我在這裡下載的
[https://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2]

aa65535, 2016-12-26:
@annbigbig 我下载过这个 SDK, 里面有上述文件, 可能是你下载的 SDK 不完整？所以建议重新下载一次。

annbigbig, 2016-12-26:
@aa65535 大俠您和我同一個地址下載的嗎？還是您可以告訴我您在哪兒下載的呢？我另外有試著編譯chinadns的ipk包，可是也失敗了，都可能是SDK缺少某些文件的關係嗎？

aa65535, 2016-12-26:
@annbigbig 是同一个地址

aa65535, 2016-12-26:
如果只是需要编译后的 ipk ，这里有一份 [ramips.zip](https://github.com/shadowsocks/openwrt-shadowsocks/files/672596/ramips.zip)

annbigbig, 2016-12-26:
@aa65535 感謝大俠相助，ramips是適用mt76xx系列的路由對嗎？我未來可能會購入linksys 1x00acs系列的路由，所以我想試著自行編譯，才是好的解決方法，我再下載一次SDK再編譯一次試試（雖然我已經試了三次以上），我發現OpenWRT官網除了SDK可以下載，還有OpenWrt-Toolchain-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2這個文件，請問編譯ipk時會需要用到它嗎？我需要下載toolchain然後解開它，並把/path/to/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/bin加到PATH環境變數嗎？

hqvv, 2016-12-26:
学习编译这类简单问题 就不要麻烦大神了吧。
建议去用别人完全配置好的虚拟机镜像。你要做的只是把包 clone进 package文件夹。然后选中ss 编译个固件，ipk自然就编译好了。什么前置工作全省了，也不可能出现什么错误。
www.right.com.cn/forum/forum.php?mod=viewthread&tid=179557&extra=page%3D4&mobile=2

annbigbig, 2016-12-26:
剛才又試著編譯了一次，結果這次是跳

```
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64$ make package/shadowsocks-libev/compile V=99 | tee build.log
tmp/.config-package.in:36:warning: ignoring type redefinition of 'PACKAGE_libc' from 'boolean' to 'tristate'
tmp/.config-package.in:64:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'boolean' to 'tristate'
#
# configuration written to .config
#
make[1]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
make[2]: Entering directory '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/feeds/base/package/libs/toolchain'
mkdir -p /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain
touch /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/.prepared_bdb409f839c046522fb7daeec3a8de40
rm -f /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp/.toolchain_installed
(cd /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/./; if [ -x ./configure ]; then find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/config.guess; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/toolchain:toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/usr/include -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/include " LDFLAGS="-L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro "   ./configure --target=mipsel [... cut ...]
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64$ 
```

我注意到這一行了
`cp: cannot stat '/home/labasky/temp/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15/lib/libgcc_s.so.*': No such file or directory`
我的SDK根目錄的staging_dir子目錄下面，根本就沒有toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.15目錄存在，所以更別說後面的/lib/libgcc_s.so.*了，請問 @aa65535 大俠我該如何是好？手動下載toolchain的tar.bz2然後放在SDK根目錄下的staging_dir子目錄裡嗎？

annbigbig, 2016-12-26:
@hqvv 我也不想麻煩各位，它少一個文件就真的沒辦法編譯，小的也自行編譯成功過數十次固件與ipk，也有用過Image Generator產生過固件，我也不知道為什麼我clone了shadowsocks和chinadns之後，怎麼樣都編譯不起來，一直報錯一直缺少文件，所以大家都是一試就成這麼簡單嗎？我沒有恩山帳戶我待會兒會注冊然後看您給的連結講的虚拟机镜像（我沒用過）是什麼，謝謝

annbigbig, 2016-12-26:
@hqvv 我支付寶被鎖死了，沒辦法付1元，看不到您指向的放在恩山論壇的教程，請問有其他的教程連結嗎？（關於您講的虚拟机镜像產生shadowsocks的ipk的方法）

aa65535, 2016-12-26:
@annbigbig 完整的 SDK 是有这个目录的，你是怎么解压的 SDK 压缩包？

annbigbig, 2016-12-26:
@aa65535 謝謝大俠，我就執行

`tar jxf ./OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2`

這樣子解壓縮的
請問這樣子會少解開什麼子目錄嗎？

aa65535, 2016-12-26:
@annbigbig 建议还是按照 [README](https://github.com/shadowsocks/openwrt-shadowsocks#编译) 重新操作一遍吧。

wongsyrone, 2016-12-26:
你下载的那个目录有checksum（sha256，md5），下载完sdk校验一下不就清楚了，sdk完整再解压

annbigbig, 2016-12-26:
我在想會不會是因為我下載的SDK是trunk (DD)的
這一個
https://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2
下載完解壓縮之後，SDK根目錄下真的沒有staging_dir/toolchain-xxxxxxx子目錄存在
我稍後試著下載Chaos Calmer 15.05.1版的再試一次
@aa65535 大俠我確定我有照著README的步驟，每一步都作確實了

wongsyrone, 2016-12-26:
trunk的sdk没意义，而且有bug，用trunk不如直接用LEDE代码树。虽然OpenWrt说自己没死，但是很久没有新提交。

aa65535, 2016-12-26:
@annbigbig 你还是用 md5sum 检查一下 文件下载的是否正确吧
```
$ md5sum OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2 
c16e2eae1c6d3e1caff2135ecce779a7  OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2
```

annbigbig, 2016-12-26:
@aa65535 @wongsyrone 好的，不好意思麻煩大家的，我真的沒那麼肉包，其實以前也常常從source code自已build成功，不知道為什麼這一次這麼不順利，我再試試看，先略過trunk的sdk，我先下載Chaos Calmer 15.05.1的自已再試試看，非常感謝各位大俠

annbigbig, 2016-12-26:
@aa65535 糟糕了，md5sum看起來和您的非常不同，該不會小的遇到傳說中的middle man attack了吧？
```
labasky@labasky-Lenovo-B590:~/temp$ md5sum ./OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2 
845c312213b97e8342b892e96f7a5b0f  ./OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2
```

aa65535, 2016-12-26:
应该是没有下载完整，通过文件大小应该也能看出来。

annbigbig, 2016-12-26:
@aa65535 感謝大俠指導，小的先下載Chaos Calmer 15.05.1版的SDK再試一次

annbigbig, 2016-12-26:
@aa65535 @wongsyrone 兩位大俠，小的剛才除了改下載Chaos Calmer 15.05.1版的SDK（下載完有驗證md5sum無誤），我發現目前我的電腦是裝Ubuntu 16.10，上面預設的gcc/g++ compiler居然是6.2版，我將它們都降級到4.8版，然後它比稍早前多跑了build process數分鐘之久，最後停在這個錯誤訊息

```
checking for mbedtls_cipher_setup in -lmbedcrypto... no
configure: error: mbed TLS libraries not found.
Makefile:83: recipe for target '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev/mbedtls/shadowsocks-libev-2.5.6/.configured_yynn' failed
make[2]: *** [/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev/mbedtls/shadowsocks-libev-2.5.6/.configured_yynn] Error 1
make[2]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/shadowsocks-libev'
package/Makefile:191: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:174: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2
```

小的覺得關鍵字可能是
configure: error: mbed TLS libraries not found.
這一行，好像快要成功了是嗎？就差臨門一腳

aa65535, 2016-12-26:
@annbigbig 

```
./scripts/feeds update base
./scripts/feeds install libpolarssl libmbedtls
rm -rf package/feeds/base/mbedtls/patches
```

annbigbig, 2016-12-26:
@aa65535 好的，謝謝大俠，小的再試試

annbigbig, 2016-12-26:
@aa65535 @wongsyrone 兩位大俠我照著README作完

```
$ ./scripts/feeds update packages
$ ./scripts/feeds install libpcre
$ ./scripts/feeds update base
$ ./scripts/feeds install libc zlib libopenssl libpolarssl libmbedtls
```

之後，我的/package/feeds/base/mbedtls/patches目錄不存在，如下
```
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/feeds/base$ pwd
/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/feeds/base
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/feeds/base$ ls -al
總計 8
drwxrwxr-x 2 labasky labasky 4096 12月 26 23:23 .
drwxrwxr-x 4 labasky labasky 4096 12月 26 23:23 ..
lrwxrwxrwx 1 labasky labasky   51 12月 26 23:23 ocf-crypto-headers -> ../../../feeds/base/package/libs/ocf-crypto-headers
lrwxrwxrwx 1 labasky labasky   40 12月 26 23:23 openssl -> ../../../feeds/base/package/libs/openssl
lrwxrwxrwx 1 labasky labasky   41 12月 26 23:23 polarssl -> ../../../feeds/base/package/libs/polarssl
lrwxrwxrwx 1 labasky labasky   42 12月 26 23:23 toolchain -> ../../../feeds/base/package/libs/toolchain
lrwxrwxrwx 1 labasky labasky   37 12月 26 23:23 zlib -> ../../../feeds/base/package/libs/zlib
```

我有執行
`rm -rf package/feeds/base/mbedtls/patches`
可是應該沒用，要刪除的子目錄根本不存在

我執行了
```
make menuconfig
make package/shadowsocks-libev/compile V=99
```

最後依然是
```
checking for thread local storage (TLS) class... __thread
checking for mbedtls_cipher_setup in -lmbedcrypto... no
configure: error: mbed TLS libraries not found.
Makefile:83: recipe for target '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev/mbedtls/shadowsocks-libev-2.5.6/.configured_yynnn' failed
make[2]: *** [/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev/mbedtls/shadowsocks-libev-2.5.6/.configured_yynnn] Error 1
make[2]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/shadowsocks-libev'
package/Makefile:191: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
/home/labasky/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:174: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

```

請問我需要把這個patch
https://github.com/wongsyrone/lede-1/blob/master/package/libs/mbedtls/patches/999-tweak-config-for-shadowsocks.patch
複製到SDK目錄裡的某個地方嗎？
還是我不需要管這個patch？
還有另外這個patch
https://github.com/wongsyrone/lede-1/blob/master/package/libs/mbedtls/patches/200-config.patch
我需要複製到SDK目錄裡的某個地方嗎？

不好意思一直麻煩您

aa65535, 2016-12-27:
@annbigbig 如果你使用 15.05.1 的 SDK ，因为它不带 base feeds, 需要手动加入 libpolarssl libmbedtls 这两个软件包的 Makefile, 或者 make menuconfig 的时候不选择编译下面这四个 ipk
```
shadowsocks-libev-mbedtls
shadowsocks-libev-server-mbedtls
shadowsocks-libev-polarssl
shadowsocks-libev-server-polarssl
```

annbigbig, 2016-12-27:
@aa65535 大俠我不知道怎麼手動加入libpolarssl libmbedtls 这两个软件包的 Makefile，不知道要從那裡下載，即使下載了，也不知道要加到SDK目錄裡的什麼地方？所以我make menuconfig之後，不要選您說的那四個ipk應該就行了吧？小的先出門工作了，下班後小的會照您指示的再試一次，非常感謝大俠您的耐心指導

annbigbig, 2016-12-27:
@aa65535 @wongsyrone 兩位大俠，請問我要怎樣，才能像你們一樣，遇到這些問題都能自行解決呢？要學c/c++然後自已clone源碼下來一行一行研究嗎？感覺我現在遇到的問題，不外乎是SDK裡少了文件、要apply patch或是不要apply patch，或是compiler版本號不正確之類的雞毛蒜皮問題，或是feeds裡少了什麼，可是我自已都沒辦法解決，git clone了Openwrt的repository下來，感覺超巨大不知道該從那裡開始，有適合新手理解Openwrt的教程嗎？(目前我雖然會編譯固件，也會用Image Generator，可是對源碼還有它的內部一無所知，有點心虛），希望將來可以自已學會抓魚，而不是一直麻煩大俠們給魚吃

wongsyrone, 2016-12-27:
这个主要是读Makefile，还有多折腾，平时不折腾临时抱佛脚肯定是不知道如何下手的。

annbigbig, 2016-12-27:
編譯終於成功了，感謝 @aa65535 @wongsyrone @hqvv 各位大俠相助

```
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ramips/packages/base$ ls -al
總計 1908
drwxr-xr-x 2 labasky labasky   4096 12月 27 17:34 .
drwxr-xr-x 4 labasky labasky   4096 12月 27 17:31 ..
-rw-r--r-- 1 labasky labasky   9470 12月 27 17:26 ldconfig_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky   5524 12月 27 17:26 ldd_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky   5590 12月 27 17:26 libatomic_4.8-linaro-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 213672 12月 27 17:26 libc_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky  31916 12月 27 17:26 libgcc_4.8-linaro-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 675991 12月 27 17:29 libopenssl_1.0.2j-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 130882 12月 27 17:30 libpolarssl_1.3.14-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky  31065 12月 27 17:26 libpthread_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky   5683 12月 27 17:26 librt_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 244319 12月 27 17:26 libstdcpp_4.8-linaro-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky   8270 12月 27 17:26 libthread-db_0.9.33.2-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 195767 12月 27 17:29 openssl-util_1.0.2j-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky 222338 12月 27 17:34 shadowsocks-libev_2.5.6-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky  92805 12月 27 17:34 shadowsocks-libev-server_2.5.6-1_ramips_24kec.ipk
-rw-r--r-- 1 labasky labasky  40277 12月 27 17:26 zlib_1.2.8-1_ramips_24kec.ipk
labasky@labasky-Lenovo-B590:~/temp/OpenWrt-SDK-15.05.1-ramips-mt7620_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ramips/packages/base$ 

```