shadowsocks/openwrt-shadowsocks #317: UDP-Relay 无法正常工作

owwo, 2017-01-29:
我用的是 NETGEAR WNDR3800 的路由，只要开启 UDP-Relay，DNS 就不能正常解析了 ( 我想这个大概是它没正常工作吧

### 安装的版本
* shadowsocks-libev_2.6.3-1_ar71xx.ipk
* luci-app-shadowsocks_1.5.0-1_all.ipk

### 设置 UDP-Relay 为 ```Same as Main Server``` 的启动日志

```
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: initializing ciphers... chacha20
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: tcp port reuse enabled
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: UDP relay enabled
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: udp port reuse enabled
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: listening at 0.0.0.0:1234
Sun Jan 29 18:42:36 2017 kern.info ss-redir[6609]: running from root user
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: initializing ciphers... chacha20
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: tcp port reuse enabled
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: udprelay enabled
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: udp port reuse enabled
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: listening at 0.0.0.0:1080
Sun Jan 29 18:42:38 2017 kern.info ss-local[6719]: running from root user
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: initializing ciphers... chacha20
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: tcp port reuse enabled
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: UDP relay enabled
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: udp port reuse enabled
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: listening at 0.0.0.0:5300
Sun Jan 29 18:42:38 2017 kern.info ss-tunnel[6748]: running from root user
```

### 设置 UDP-Relay 为 ```Disable``` 的启动日志

```
Sun Jan 29 18:44:26 2017 kern.info ss-redir[6842]: initializing ciphers... chacha20
Sun Jan 29 18:44:26 2017 kern.info ss-redir[6842]: tcp port reuse enabled
Sun Jan 29 18:44:26 2017 kern.info ss-redir[6842]: listening at 0.0.0.0:1234
Sun Jan 29 18:44:26 2017 kern.info ss-redir[6842]: running from root user
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: initializing ciphers... chacha20
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: tcp port reuse enabled
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: udprelay enabled
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: udp port reuse enabled
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: listening at 0.0.0.0:1080
Sun Jan 29 18:44:28 2017 kern.info ss-local[6943]: running from root user
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: initializing ciphers... chacha20
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: tcp port reuse enabled
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: UDP relay enabled
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: udp port reuse enabled
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: listening at 0.0.0.0:5300
Sun Jan 29 18:44:28 2017 kern.info ss-tunnel[6972]: running from root user
```

我明明设置 UDP-Relay 为 Disable 为啥日志这里还是 ```UDP relay enabled``` ？

### 其他问题

 ```Access Control -> Zone LAN -> Interface``` 选项不能取消

aa65535, 2017-02-02:
- Sun Jan 29 18:44:28 2017 kern.info **ss-tunnel**[6972]: UDP relay enabled
   这里是 ss-tunnel 不是 ss-redir

 - Interface 必须有至少一个选中，留空代表选中所有

closed 2017-02-02 by aa65535

owwo, 2017-02-02:
@aa65535 但是为什么我使用 luci-app-shadowsocks v1.3.7 以上版本 UDP-Relay 无法正常开启了？