shadowsocks/openwrt-shadowsocks #323: 请教iptables-mod-tproxy模块安装问题

smthvitas, 2017-02-03:
刷的merlin，装的entware，发现repository中间没有iptables-mod-tproxy.
试着改了下entware的源，直接指向openwrt 博通方案的源：
http://downloads.openwrt.org/snapshots/trunk/bcm53xx/generic/packages/base/
也不行，提示架构不匹配。

请教：
1、这种情况下安装tproxy模块最便捷的方法是不是卸载entware，重装optware再通过opkg安装？
2、如果希望把iptables-mod-tproxy移植到entware环境，是单纯编译一下就可以，还是需要C语言开发背景？如果是前者，能否推荐一些学习资料？

谢谢！

legendchi, 2017-02-19:
拿到SDK单独编译一下iptables-mod-tproxy就好了，

xiasonming, 2017-03-11:
make menuconfig
Network--->Firewall--->iptables--->iptables-mod-tproxy  直接勾选，再编译

Portan, 2017-03-24:
@legendchi 能具体说下怎么编译吗？

kohillyang, 2017-04-08:
Please refer to this site:<http://www.jianshu.com/p/1042483f90fe>

kohillyang, 2017-04-08:
choose iptables-mod-tproxy in kernel modules-network netfilter when make menuconfig

closed 2017-05-06 by aa65535