shadowsocks/openwrt-shadowsocks #325: 编译报错，请求帮助

xingezhang, 2017-02-06:
出错信息如下：

Hunk #1 FAILED at 4.
Hunk #2 succeeded at 74 (offset 6 lines).
Hunk #3 succeeded at 100 (offset 6 lines).
1 out of 3 hunks FAILED -- saving rejects to file src/Makefile.am.rej
Patch failed!  Please fix ./patches/0001-embed-libudns.patch!
Makefile:62: recipe for target '/home/xinge/ar71xx-sdk/build_dir/target-mips_24kc_musl-1.1.15/shadowsocks-libev//shadowsocks-libev-3.0.0-1/.prepared_046b412e71f3b21f3b62fada20876e99' failed
make[2]: *** [/home/xinge/ar71xx-sdk/build_dir/target-mips_24kc_musl-1.1.15/shadowsocks-libev//shadowsocks-libev-3.0.0-1/.prepared_046b412e71f3b21f3b62fada20876e99] Error 1
make[2]: Leaving directory '/home/xinge/ar71xx-sdk/package/shadowsocks-libev'
package/Makefile:105: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/xinge/ar71xx-sdk'
/home/xinge/ar71xx-sdk/include/toplevel.mk:190: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

哪位有时间帮忙看看，谢谢！

xingezhang, 2017-02-06:
更新：
删除package/shadowsocks-libev/patch/0001-embed-libudns.patch

 再次编译，提示找不到mbedtls，下载https://github.com/lede-project/source/tree/master/package/libs/mbedtls 放在package下后

再次编译：
checking udns.h usability... no
checking udns.h presence... no
checking for udns.h... no
configure: error: Couldn't find libudns. Try installing libudns-dev or udns-devel.

目前被卡在这里，不知道这个该怎么解决

xingezhang, 2017-02-06:
仔细看了好像0001-embed-libudns.patch不能删，但是如果不删的话，会一直提示Patch failed! Please fix ./patches/0001-embed-libudns.patch!

这个应该怎么破？

xingezhang, 2017-02-06:
我猜测这个0001-embed-libudns.patch和libudns有关系，但是为什么一直说有问题呢？

aa65535, 2017-02-06:
是否有修改 Makefile ?

xingezhang, 2017-02-06:
@aa65535 感谢指点，全部清空，重新clone，检查确认是833594e后，编译成功。那么再请教一下，如果想用最新的commit编译的话，要怎么做呢？

xingezhang, 2017-02-06:
再次感谢@65535,我摸索出来怎么给最新版打patch了，最新commit也已经编译成功。

closed 2017-02-06 by xingezhang