shadowsocks/openwrt-shadowsocks #326: 编译出错，求助

tyadan, 2017-02-06:
configure: error: Invalid kernel source directory /lib/modules/4.8.0-37-generic/source
Makefile:60: recipe for target '/home/jayshao/source/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/ipset-6.30/.configured_yyyyyyy' failed
make[2]: *** [/home/jayshao/source/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/ipset-6.30/.configured_yyyyyyy] Error 1
make[2]: Leaving directory '/home/jayshao/source/package/network/utils/ipset'
package/Makefile:106: recipe for target 'package/network/utils/ipset/compile' failed
make[1]: *** [package/network/utils/ipset/compile] Error 2
make[1]: Leaving directory '/home/jayshao/source'
/home/jayshao/source/include/toplevel.mk:197: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

编译环境基于最新的 Lede Reboot 分支

xingezhang, 2017-02-06:
看这句make[2]: *** [/home/jayshao/source/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/ipset-6.30/.configured_yyyyyyy] Error 1
好象是ipset有问题

cokebar, 2017-02-06:
configure: error: Invalid kernel source directory /lib/modules/4.8.0-37-generic/source
内核源码目录错误
话说LEDE不是4.4.42内核么

closed 2017-02-07 by aa65535