shadowsocks/openwrt-shadowsocks #334: UDP转发

caoyang22722, 2017-02-13:
请问大神，下图是我自建VPS服务器上的截图：
[image]

是不是说明服务器端已经开启了SS服务？包括UDP转发？

如果服务器已经转发设置完成，请问下在openwrt上如何开启带UDP转发的SS？（游戏的UDP转发）
刚刷完openwrt后，自带一个ss软件，但是没有udp转发功能的，不过可以通过上面的VPS的SS服务翻墙；
但是刷入带udp转发的ss后（2.4.8），翻墙也不行了，youtube打不开，但是开全局模式打开ip138.com后ip显示的是我vps的ip，也不知道udp转发成功没，请问我该怎么解决？请大神指教~

caoyang22722, 2017-02-13:
补充一下，openwrt上的ss（2.4.8）是从这里下的：
https://sourceforge.net/projects/openwrt-dist/files/shadowsocks-libev/

作者说后续转移到了
https://bintray.com/aa65535/opkg/shadowsocks-libev

但是2.4之后好像就没有spec版本的可以下载了吗？最新版本的怎么才能找到spec（可以结合luci的那种图形化）版本呢？

aa65535, 2017-02-13:
v2.4.8-3 以后搭配 https://github.com/shadowsocks/luci-app-shadowsocks 使用就可以了，具体说明参考 [WIKI](https://github.com/shadowsocks/luci-app-shadowsocks/wiki).

closed 2017-02-13 by aa65535