shadowsocks/openwrt-shadowsocks #336: 能不能编译一个armv7的版本？

fyhong, 2017-02-18:
我的设备是armv7的。
https://dl.bintray.com/aa65535/opkg/shadowsocks-libev/3.0.2/OpenWrt/ramips/
到这里所有版本都下过了。没有一个能正常运行。。大神能不能帮忙编译一个armv7的版本

aa65535, 2017-02-19:
设备型号？
OpenWrt版本？

fyhong, 2017-02-20:
不是openwrt的行不行。。
telnet 125.214.234.108:30819
你登上我的shell看看。是监控设备来的。。这是cpu信息. 网上找的1.6.1可以用。

processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 2190.54
Features        : swp half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : bigfish
Revision        : 0000
Serial          : 0000000000000000

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

aa65535, 2017-02-20:
不是 OpenWrt/LEDE 的话无能为力，可以去找找官方有没有提供 SDK 或者 Toolchain，然后尝试编译.

closed 2017-02-20 by aa65535