shadowsocks/openwrt-shadowsocks #337: 编译遇到了问题

daiaji, 2017-02-19:
```
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for thread local storage (TLS) class... __thread
checking for mbedtls_cipher_setup in -lmbedcrypto... no
configure: error: mbed TLS libraries not found.
Makefile:62: recipe for target '/home/homo/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev//shadowsocks-libev-3.0.2-1/.configured_nnnnnnnn' failed
make[2]: *** [/home/homo/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/shadowsocks-libev//shadowsocks-libev-3.0.2-1/.configured_nnnnnnnn] Error 1
make[2]: Leaving directory '/home/homo/openwrt/package/shadowsocks-libev'
package/Makefile:191: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/homo/openwrt'
/home/homo/openwrt/include/toplevel.mk:181: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2
```
以及
` ./scripts/feeds install zlib libev libsodium libopenssl libmbedtls libpcre`
的时候反馈
`WARNING: No feed for package 'libmbedtls' found, maybe it's already part of the standard packages?`
编译Openwrt 15.05.1 Chaos Calmer

aa65535, 2017-02-19:
```
git clone https://github.com/aa65535/openwrt-feeds.git package/feeds
```

closed 2017-02-19 by aa65535

daiaji, 2017-02-19:
homo@homo-VirtualBox:~/OpenWrt-SDK-15.05.1-ramips-rt288x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$ git clone https://github.com/aa65535/openwrt-feeds.git package/feeds
正克隆到 'package/feeds'...
remote: Counting objects: 35, done.
remote: Total 35 (delta 0), reused 0 (delta 0), pack-reused 35
展开对象中: 100% (35/35), 完成.
检查连接... 完成。
homo@homo-VirtualBox:~/OpenWrt-SDK-15.05.1-ramips-rt288x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$ make menuconfig
make -s -C scripts/config all CC=cc: build failed. Please re-run make with -j1 V=s to see what's going on
/home/homo/OpenWrt-SDK-15.05.1-ramips-rt288x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:89: recipe for target 'scripts/config/mconf' failed
make: *** [scripts/config/mconf] Error 1
homo@homo-VirtualBox:~/OpenWrt-SDK-15.05.1-ramips-rt288x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$ 

我可能下了假的SDK

aa65535, 2017-02-20:
@daiaji `make menuconfig` 都过不了的话，检查一下 SDK 是否下载完整了。

daiaji, 2017-02-20:
我跳过了`make menuconfig` 
执行这个还是会
```
/home/homo/OpenWrt-SDK-15.05-ramips-rt288x_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:89: recipe for target 'scripts/config/mconf' failed
make: *** [scripts/config/mconf] Error 1
```
直接`make package/shadowsocks-libev/compile V=99`
然后没有报错
这样没问题吗？
检验了SDK的压缩包 没问题
但是很奇怪感觉是openwrt那边把SDK换了版本15.05.1变成了15.05