shadowsocks/openwrt-shadowsocks #342: Error out after running `make menuconfig`

shengyang998, 2017-02-23:
I did download the sdk, clone the feeds and the Makefile, but after running command `make menuconfig` , it occured an error: 
```shell
make -s -C scripts/config all CC=/Volumes/OpenWrt/OpenWrt-SDK-15.05.1-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/scripts/clang-gcc-wrapper: build failed. Please re-run make with -j1 V=s to see what's going on
make: *** [scripts/config/mconf] Error 1
```
Why is that? Is that any tool I didn't install? How to fix it? Or should I paste the result of running `make -j1 -V=s menuconfig`?

BTW, my operating system is macOS Sierra. 
And I also did as [this post](https://wiki.openwrt.org/inbox/easy.build.macosx) and [this post](https://wiki.openwrt.org/doc/howto/buildroot.exigence.macosx) said.

lixingcong, 2017-02-23:
Please re-run make with -j1 V=s to see what's going on

It will take about ten minutes.

aa65535, 2017-02-23:
https://lede-project.org/docs/guide-developer/install-buildsystem

closed 2017-02-23 by aa65535