shadowsocks/openwrt-shadowsocks #343: compiling error

debiansid, 2017-02-23:
I update to newest trunk of your package,and installed openwrt-udns, but still got this error

I doubt it was because openwrt-udns named libudns as udns-libs.
would you please check on this?

thanks

```Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev:
 *      libudns *
 * opkg_install_cmd: Cannot install package shadowsocks-libev.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/storage/lede-gw-64'
make[1]: *** [/home/storage/lede-gw-64/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/storage/lede-gw-64'
make: *** [world] Error 2
```

debiansid, 2017-02-23:
I tried to rename udns-libs to libudns, now it works.

aa65535, 2017-02-23:
···
git clone https://github.com/shadowsocks/openwrt-feeds.git package/feeds
···

https://github.com/shadowsocks/openwrt-feeds/blob/master/packages/libudns/Makefile#L64

closed 2017-02-23 by aa65535