shadowsocks/openwrt-shadowsocks #355: Failed when check "Ignore resolve file" at "DHCP and DNS"

yukeiyang, 2017-03-13:
**Router:** 
Xiaomi Mini

**Openwrt:** 
15.05

**Shadowsocks-libev:**
Newest version by compile 

**LUCI:**
All Newest version,  for Shadowsocks-libev, ChinaDNS, Shadowsocks, DNSForwarder

**Simple-obfs:** 
v0.0.3

ShadowSocks - Access Control - Bypassed IP List - Choose "ChinaDNS-CHNROUTE"

DHCP and DNS - DNS forwardings - Filled with "0.0.0.0#5353"

But after check "Ignore resolve file" for /tmp/resolv.conf.auto, SS will failed to access the internet.
Can u give me some suggestions?

aa65535, 2017-03-14:
```sh
echo 'search lan' >/etc/resolv.conf
echo 'nameserver 127.0.0.1' >>/etc/resolv.conf
```

yukeiyang, 2017-03-14:
**@aa65535 Thanks for your help. In my case, Success after these steps below:**

**1. ShadowSocks - General Settings - Port Forward**
       - Local Port - 5300
       - Destination - 8.8.4.4:53

**2. ShadowSocks - Access Control - Zone LAN **
       - Proxy Type - Nomal
       - Self Proxy - Global
 
**3. DNS-Forwarder**     
       - Listen Port - 5301 

**4. ChinaDNS**
       - Upstream Servers - 114.114.114.114,127.0.0.1:5301

**5. DHCP and DNS - General Settings** 
       - DNS forwardings - 127.0.0.1#5353
       - Resolv and Hosts Files - Ignore resolve file  (check it)
                                              - Ignore Hosts files   (check it)

closed 2017-03-14 by aa65535