shadowsocks/openwrt-shadowsocks #357: 安装后无法启动

xingezhang, 2017-03-15:
EA4500 刷LEDE 17.0.1
可以正常上网，安装shadowsocks-libev3.0.3-1，无法启动提示如下：
[root@LEDE /etc Wed Mar 15/13:26]
$/etc/init.d/shadowsocks restart
/etc/rc.common: line 145: ss-redir: Permission denied

求解救

aa65535, 2017-03-15:
安装了错误的软件包。

xingezhang, 2017-03-15:
可是我下载的就是kirkwood / arm-xscale版本的啊

aa65535, 2017-03-15:
提供一下以下命令打印的结果
```sh
ls -l $(dirname $(which ss-redir)) | grep ss-
```

xingezhang, 2017-03-15:
设备不在手边，明天回家立即发上来

xingezhang, 2017-03-16:
[root@LEDE ~ Wed Mar 15/13:25]
$ls -l $(dirname $(which ss-redir)) | grep ss-
BusyBox v1.25.1 () multi-call binary.

Usage: dirname FILENAME

Strip non-directory suffix from FILENAME

aa65535, 2017-03-16:
@xingezhang shadowsocks-libev 是怎么安装的？

`which ss-redir` 有打印结果吗？

xingezhang, 2017-03-16:
我自己下载了sdk编译，可以安装，但是无法浏览，浏览器提示The client and server don't support a common SSL protocol version or cipher suite，
很奇怪的是我的另外一台wndr4300就没有任何问题，不管是用你提供的还是我自己编译的，都没有问题

xingezhang, 2017-03-16:
@aa65535 scp 上传到路由器，然后ssh登录，使用opkg install shadowsocks-libev libmbedtls udns完成安装的

aa65535, 2017-03-16:
`Permission denied` 一般是因为文件没有可执行权限，检查一下是不是在其他位置还有 ss 的二进制文件存在。
所以检查 `which ss-redir` 返回的结果是不是 `/usr/bin/ss-redir` 。

xingezhang, 2017-03-16:
@aa65535 重新刷了固件，干净的安装后，可以运行，但是无法打开网站，详情请看截图
[image]

[image]

xingezhang, 2017-03-16:
server没有任何做任何调整，wndr4300（ar71xx）没有任何问题，ea4500（arm_xscale）却无法连接，很奇怪的问题。

aa65535, 2017-03-16:
我曾经在 EA4500 上测试过，只能使用 ss 的全局模式，不经过 ss 的站点无法连接，应该是固件的防火墙的问题。

我没有尝试去解决这个问题，你可以试试在防火墙里把 LAN -> WAN  区域的 「IP动态伪装」打开。

closed 2017-03-29 by aa65535