shadowsocks/openwrt-shadowsocks #363: v3.0.4 AEAD chiphers do not work

niclau, 2017-03-19:
Hi,
I install v3.0.4 via your respository:
3.0.4/OpenWrt/ar71xx/
My device is mw4530r using offcical Openwrt 15.05.1.

All related packages are installed.
Using the depreated ciphers, everything is fine.

However, if I implemented one of the AEAD ciphers, ss-redir,local and tunnel will crash and return an error message like this:
ss-local -s ss.example.net -p 88 -b 0.0.0.0 -l 8888 -k password -m chacha20-ietf-poly1305 -v
 2017-03-19 12:13:22 INFO: initializing ciphers... chacha20-ietf-poly1305
 2017-03-19 12:13:24 INFO: listening at 0.0.0.0:8888
 2017-03-19 12:13:24 INFO: running from root user
 2017-03-19 12:13:44 INFO: connect to www.google.com:443
ss-local: can't resolve symbol 'crypto_aead_chacha20poly1305_ietf_encrypt' in lib 'ss-local'.

Can you give me some ideas?

suikatomoki, 2017-03-19:
libsodium installed?

niclau, 2017-03-19:
Yes.
root@OpenWrt:~# opkg list-installed | grep shadowsocks
shadowsocks-libev - 3.0.4-1
root@OpenWrt:~# opkg list-installed | grep libsodium
libsodium - 1.0.2-1
root@OpenWrt:~# opkg list-installed | grep libmbedtls
libmbedtls - 2.4.2-1

suikatomoki, 2017-03-19:
libsodium ver too old
try
http://openwrt-dist.sourceforge.net/packages/OpenWrt/base/ar71xx/libsodium_1.0.12-1_ar71xx.ipk

niclau, 2017-03-19:
Thank you.
It is working now.

closed 2017-03-19 by niclau