shadowsocks/openwrt-shadowsocks #364: 编译错误，错误提示如下，这两个文件系统是存在的，谢谢。

ewangsoft, 2017-03-25:
find /home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/udns-0.4/ipkg-ramips/libudns -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package libudns is missing dependencies for the following libraries:
libc.so
libgcc_s.so.1
Makefile:64: recipe for target '/home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64/bin/ramips/packages/base/libudns_0.4-1_ramips.ipk' failed
make[2]: *** [/home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64/bin/ramips/packages/base/libudns_0.4-1_ramips.ipk] Error 1
make[2]: Leaving directory '/home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64/package/feeds/base/libudns'
package/Makefile:196: recipe for target 'package/feeds/base/libudns/compile' failed
make[1]: *** [package/feeds/base/libudns/compile] Error 2
make[1]: Leaving directory '/home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64'
/home/wang/OpenWrt-SDK-ramips-mt7620_gcc-5.3.0_musl-1.1.16.Linux-x86_64/include/toplevel.mk:185: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2

ewangsoft, 2017-03-25:
$ uname -a
Linux ubuntuserver 4.4.0-46-generic #67-Ubuntu SMP Thu Oct 20 15:05:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

去downloads.openwrt.org下载的RAMIPS的SDK，编译过程参照的是本项目的readme

aa65535, 2017-03-26:
建议使用 LEDE 的 SDK 编译， trunk 版本的 SDK 可能会有一些问题， libc 和 libgcc_s 是指 toolchain 里的，不是系统的。

正确的 SDK 里应该有 `package/toolchain` 这个目录。

closed 2017-03-26 by aa65535