shadowsocks/openwrt-shadowsocks #369: Bintray repo

fanthos, 2017-04-16:
We can downlaod binary from bintray.com directly. Bintray also provide a service that we can use it as opkg repo. 
Right now opkg works if I added `src/gz ss https://dl.bintray.com/aa65535/opkg/shadowsocks-libev/3.0.5/OpenWrt/x86` to the opkg configuration file. But if I want to update the package with opkg, I have to change the version number in the opkg configuration file.

Can you also upload a copy of released file to a folder without version number like `https://bintray.com/aa65535/opkg/shadowsocks-libev/lastest`, so I can update my package using opkg without manually change the repo url?

aa65535, 2017-04-16:
https://github.com/shadowsocks/openwrt-shadowsocks/issues/304

closed 2017-04-16 by aa65535

fanthos, 2017-04-16:
The sf site didn't update since version 2.4.8.

?? Outlook for Android<https://aka.ms/ghei36>

#304<https://github.com/shadowsocks/openwrt-shadowsocks/issues/304>

aa65535, 2017-04-16:
http://openwrt-dist.sourceforge.net/packages/