shadowsocks/openwrt-shadowsocks #371: x86_64 upgrade,  has no valid architecture, ignoring

mysteriouss, 2017-04-19:
```
Package ChinaDNS version 1.3.2-4 has no valid architecture, ignoring.
Package ShadowVPN version 0.2.0-1 has no valid architecture, ignoring.
Package dns-forwarder version 1.2.0-1 has no valid architecture, ignoring.
Package libmbedtls version 2.4.2-1 has no valid architecture, ignoring.
Package libsodium version 1.0.12-1 has no valid architecture, ignoring.
Package libudns version 0.4-1 has no valid architecture, ignoring.
Package shadowsocks-libev-server version 3.0.5-1 has no valid architecture, ignoring.
Package shadowsocks-libev version 3.0.5-1 has no valid architecture, ignoring.
Package simple-obfs-server version 0.0.3-1 has no valid architecture, ignoring.
Package simple-obfs version 0.0.3-1 has no valid architecture, ignoring.
Upgrading luci-app-dns-forwarder on root from 1.6.0-1 to 1.6.1-1...
Downloading http://openwrt-dist.sourceforge.net/packages/OpenWrt/luci/luci-app-dns-forwarder_1.6.1-1_all.ipk.
Configuring luci-app-dns-forwarder.
Package ChinaDNS version 1.3.2-4 has no valid architecture, ignoring.
Package ShadowVPN version 0.2.0-1 has no valid architecture, ignoring.
Package dns-forwarder version 1.2.0-1 has no valid architecture, ignoring.
Package libmbedtls version 2.4.2-1 has no valid architecture, ignoring.
Package libsodium version 1.0.12-1 has no valid architecture, ignoring.
Package libudns version 0.4-1 has no valid architecture, ignoring.
Package shadowsocks-libev-server version 3.0.5-1 has no valid architecture, ignoring.
Package shadowsocks-libev version 3.0.5-1 has no valid architecture, ignoring.
Package simple-obfs-server version 0.0.3-1 has no valid architecture, ignoring.
Package simple-obfs version 0.0.3-1 has no valid architecture, ignoring.
Upgrading luci-app-shadowsocks on root from 1.3.7-1 to 1.6.3-1...
Downloading http://openwrt-dist.sourceforge.net/packages/OpenWrt/luci/luci-app-shadowsocks_1.6.3-1_all.ipk.
Configuring luci-app-shadowsocks.
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/shadowsocks is different from the conffile in the new package. The new conffile will be placed at /etc/config/shadowsocks-opkg.
```

```
root@OpenWrt:~# opkg print-architecture shadowsocks-libev
arch all 1
arch noarch 1
arch x86_64 10
```

Thanks.

aa65535, 2017-04-20:
try again.

mysteriouss, 2017-04-20:
ok, works now. thanks

closed 2017-04-20 by mysteriouss