shadowsocks/openwrt-shadowsocks #375: 添加你的软件源后貌似无法解决和LEDE官方软件源中的包的冲突

cokebar, 2017-05-31:
LEDE官方源中的版本也更到了3.0.6，包名、版本号和你的包完全一致，opkg install/update shadowsocks-libev会导致直接装上LEDE官方软件源的包，而不是第三方源的。

不知道是否有办法解决。

tedaz, 2017-06-02:
@cokebar 我的办法是卸载feeds中的shadowsocks
`./scripts/feeds uninstall shadowsocks-client
./scripts/feeds uninstall shadowsocks-libev
./scripts/feeds uninstall shadowsocks-libev-polarssl

./scripts/feeds uninstall luci-app-shadowsocks-libev
./scripts/feeds uninstall luci-i18n-shadowsocks-libev-sv
./scripts/feeds uninstall luci-i18n-shadowsocks-libev-zh-cn`

cokebar, 2017-06-02:
编译的时候还好。
以前可以添加openwrtdist.sourceforge到opkg的源里，直接opkg install shadowsocks-libev安装，升级。
但是现在不行了，这样会安装官方源的包。
我就想问问有办法让第三方源优先么。

suikatomoki, 2017-06-06:
https://downloads.lede-project.org/releases/17.01.1/packages/mips_24kc/packages/

[image]

我怎么看到还是2.6.1

Biercenter, 2017-06-06:
> packages/mips_24kc/packages/

lede安装脚本，一键搞定
wget http://openwrt-dist.sourceforge.net/auto_install_lede.sh
chmod +x auto_install_lede.sh
./auto_install_lede.sh

cokebar, 2017-06-06:
@suikatomoki snapshot中已经是3.0.6
@Biercenter You haven't get my idea.

suikatomoki, 2017-06-06:
@cokebar 看了一下的确是这样 而且连ss-local都没有
[image]

@yousong  可以修复一下吗

suikatomoki, 2017-06-06:
ss-rules 不是 luci 包里的东西吗 越看越不对劲

cokebar, 2017-06-06:
@suikatomoki lede官方的包和aa65535的不同，还是老的结构

yousong, 2017-06-06:
> @cokebar 看了一下的确是这样 而且连ss-local都没有

看一下git历史，OpenWrt/LEDE仓库里的shadowsocks-libev一开始就没有把ss-local打包进去。

这种把所有内容包在一个shadowsocks-libev.ipk里，我以为是不符合OpenWrt/LEDE这种发行版的特点的。我自己也不喜欢这种打包方式，所以当时自己也写了一个 [1]，配合另外一个自己写的omysocks [2]，也一直用到现在。

但是一旦进入正式的仓库，再替换掉就有历史成本了，因为在OpenWrt/LEDE里，向后兼容非常重要 [3]，除非一不小心无意为之，或者真是别无他法

 [1] https://github.com/yousong/waller/blob/master/shadowsocks-libev/Makefile
 [2] https://github.com/yousong/waller/tree/master/omysocks
 [3] https://github.com/openwrt/packages/pull/3138#issuecomment-244654321

hlwanghl, 2017-06-14:
@cokebar sourceforge是这个地址吗？https://sourceforge.net/projects/openwrt-dist/files/shadowsocks-libev/ 好像最新的还是2.4.8啊

aa65535, 2017-06-29:
冲突没办法，不过安装的时候是高版本优先，然后是按照 package feeds 顺序安装。

closed 2017-07-02 by cokebar