shadowsocks/openwrt-shadowsocks #377: 编译依赖问题

LoveyCat, 2017-06-16:
在CentOS7下编译Openwrt的IPK包，已经安装了相关的依赖，例如libmbedtls-devel和libmbedtls.
但是在编译过程中还是会出错，提示找不到libmbedtls

checking whether mips-openwrt-linux-uclibc-gcc understands -c and -o together... (cached) yes
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for thread local storage (TLS) class... __thread
checking for mbedtls_cipher_setup in -lmbedcrypto... no
configure: error: mbed TLS libraries not found.
请问下这种情况如何解决呢

```
# Installation of MbedTLS
export MBEDTLS_VER=2.4.2
wget https://tls.mbed.org/download/mbedtls-$MBEDTLS_VER-gpl.tgz
tar xvf mbedtls-$MBEDTLS_VER-gpl.tgz
pushd mbedtls-$MBEDTLS_VER
make SHARED=1 CFLAGS=-fPIC
sudo make DESTDIR=/usr install
popd
sudo ldconfig
```
通过这个方法，在/usr/lib下找到了mbedtls的库，本来以为能通过编译了，没想到下载完libev后又出现这个报错了

aa65535, 2017-06-19:
```
# 添加 feeds
git clone https://github.com/shadowsocks/openwrt-feeds.git package/feeds
```

wtlusvm, 2017-06-19:
@aa65535 在 openwrt/feeds.conf 里加入

```
src-git shadowsocks https://github.com/shadowsocks/openwrt-feeds.git
```

也出现这个错误，必须直接 git clone 到 package/feeds 吗
PS：我直接拉取了 OpenWRT-15.05 的代码

LoveyCat, 2017-06-19:
@aa65535 感谢回复！不是在SDK下面编译的话直接添加feeds确实会报错，因为package目录下已经有feeds目录了，我把他git clone 到 package/feeds/ss 然后把base目录move到package/feeds,因为剩下的两个库文件op的源码已经有了，所以就留在ss目录下了直接用Op自带的，这样的话编译就没有问题了  @wtlusvm

wtlusvm, 2017-06-20:
@LoveyCat 是的昨天我也稍微捯饬了一下，直接加到 feeds.conf 也不行（尽管有软链接），现在编译通过，只是别的软件编译出错了233

closed 2017-06-20 by LoveyCat