shadowsocks/openwrt-shadowsocks #379: 下载特定版本的S-S的SOURCE CODE后通过OPENWRT SDK 如何编译?

tacuar, 2017-06-28:
目前根本README的IPK编译流程, 可以编译成功,  比如 编译S-S,最新的make package/shadowsocks-libev/compile V=99后就可以.

但是如果下载一个特定的版本比如2.4.8, 里面有个SPEC版的, 下载后解压到package后, 怎样才能编译成功?

其中一个是SS-SPEC, 另外一个是SPEC版的LUCI

tacuar, 2017-06-28:
直接将一个1.5.0的LUCI 及 2.4.8的SPEC 放入package后, make menuconfig可以看到有, 但是最后的make package完整的应该如何写才能生成自己想要的特定版本的ipk?

aa65535, 2017-06-28:
直接在源码里 `git checkout v2.4.8` 编译就行了。

tacuar, 2017-06-28:
AA大神, 可否详细说下,  sorry.....

tacuar, 2017-06-28:
具体应该..... 比方说现在SDK的目录名是 openwrt-sdk, 然后 cd openwrt-sdk ,  package下面有我下载的2.4.8-2的, 然后如何操作呢?

tacuar, 2017-06-29:
论坛问了有的说要修改makefile的版本号, 有的说 要修改PKG_SOURCE_VERSION:=commit对应的那串代码, 请问A大, 到底如何操作呢,

tacuar, 2017-06-29:
really will need a README file for those who wants to compile versions from those archived source codes , Lord.....................

aa65535, 2017-06-29:
```sh
cd openwrt-sdk
git clone https://github.com/shadowsocks/openwrt-feeds.git package/feeds
# 获取 shadowsocks-libev Makefile
git clone https://github.com/shadowsocks/openwrt-shadowsocks.git package/shadowsocks-libev
# 切换版本
pushd package/shadowsocks-libev
git checkout v2.4.8
popd
# 选择要编译的包 Network -> shadowsocks-libev
make menuconfig
# 开始编译
make package/shadowsocks-libev/compile V=99
```

tacuar, 2017-06-29:
Thanking you really really much Lord ! Will try it out immediately!

tacuar, 2017-06-29:
by the way that verion more exactly should be v2.4.8-2, is that correct?

tacuar, 2017-06-29:
sourceforge.net has a compiled ipk luci-app-shadowsocks-spec_1.5.0-1_all.ipk, it is SPEC verion LUCI,  if i use git checkout command, could you direct me a correct branch? i could not find a SPEC version in 1.5.0-1

aa65535, 2017-06-29:
https://github.com/aa65535/openwrt-dist-luci/releases/tag/v1.5.0

tacuar, 2017-06-29:
lol, no wonder i couldn't manage to find it  : ) it is hiding so deeply in the disk-luci category................................
well, now i think the problem resolved.... 
Many thanks AA LORD !

tacuar, 2017-06-29:
The compile went on well and got all the ipks output, however it's weird that the luci still can't be installed on BB 1407 (poscript error), but all ipks are generated by the BB 1407 SDK .

Anyway, i got the official method of compiling those archived versions of source codes, still very satisfied, maybe i should simply switch to CC or LEDE and get rid of the BB.

tacuar, 2017-06-29:
Updates:  Tried several times and found the facts, the compiled ipk ss-spec and luci , even can not be packed into the BB as a whole, but they can later be  installed by opkg without any problem when the BB is flashed into the router.

closed 2017-07-07 by aa65535