shadowsocks/openwrt-shadowsocks #383: Compile failed - *** No targets specified and no makefile found.  Stop.

tedaz, 2017-07-13:
`make[3]: Entering directory '/lede/build_dir/target-x86_64_musl/shadowsocks-libev/shadowsocks-libev-3.0.7-c11b2c08e4aecaa01f7bc4e93d311f9945d84363'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory '/lede/build_dir/target-x86_64_musl/shadowsocks-libev/shadowsocks-libev-3.0.7-c11b2c08e4aecaa01f7bc4e93d311f9945d84363'
Makefile:62: recipe for target '/lede/build_dir/target-x86_64_musl/shadowsocks-libev//shadowsocks-libev-3.0.7-c11b2c08e4aecaa01f7bc4e93d311f9945d84363/.built' failed
make[2]: *** [/lede/build_dir/target-x86_64_musl/shadowsocks-libev//shadowsocks-libev-3.0.7-c11b2c08e4aecaa01f7bc4e93d311f9945d84363/.built] Error 2
make[2]: Leaving directory '/lede/package/shadowsocks-libev'
package/Makefile:109: recipe for target 'package/shadowsocks-libev/compile' failed
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/ted/volume/lede'
/home/ted/volume/lede/include/toplevel.mk:207: recipe for target 'package/shadowsocks-libev/compile' failed
make: *** [package/shadowsocks-libev/compile] Error 2
`

aa65535, 2017-07-13:
`rm -f dl/shadowsocks-libev-*`

tedaz, 2017-07-13:
Resolved, thanks for your help.

closed 2017-07-19 by aa65535