shadowsocks/openwrt-shadowsocks #385: 编译出错

crackself, 2017-07-19:
Openwrt官方SDK编译，以下是输出错误：

`crackself@ubuntu:~/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$ make -j1 package/shadowsocks-libev/compile V=99
#
# configuration written to .config
#
make[1]: Entering directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
make[2]: Entering directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/feeds/base/libudns'
make -C /home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/udns-0.4 all
make[3]: Entering directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/udns-0.4'
"ccache_cc" -O -DHAVE_CONFIG_H -c udns_dn.c
/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/host/bin/ccache_cc: 2: exec: ccache: not found
make[3]: *** [udns_dn.o] Error 127
make[3]: Leaving directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/udns-0.4'
make[2]: *** [/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mips_34kc_uClibc-0.9.33.2/udns-0.4/.built] Error 2
make[2]: Leaving directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/feeds/base/libudns'
make[1]: *** [package/feeds/base/libudns/compile] Error 2
make[1]: Leaving directory `/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
make: *** [package/shadowsocks-libev/compile] Error 2
crackself@ubuntu:~/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$`

aa65535, 2017-07-19:
```
/home/crackself/SDK/OpenWrt-SDK-15.05.1-ar71xx-nand_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/host/bin/ccache_cc: 2: exec: ccache: not found
```
安装 ccache 即可。

closed 2017-07-19 by aa65535

crackself, 2017-07-19:
谢谢，已解决。