shadowsocks/openwrt-shadowsocks #388: Make menuconfig 出错，求指点

CXOldStar, 2017-07-28:
- 下载好github的代码后，进行make menuconfig时候出错，错误信息如下：
> root@LEDE:/tmp/sdk# make menuconfig
> /bin/sh: tty: not found
> /bin/sh: tty: not found
> make -s -C scripts/config all CC=cc: build failed. Please re-run make with -j1 V=s to see what's going on
> make: *** [/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:103: scripts/config/mconf] Error 1

- 根据提示, 执行 make -j1 V=s后的错误如下：
> root@LEDE:/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64# make -j1 V=s
> /bin/sh: tty: not found
/bin/sh: tty: not found
/bin/sh: tty: not found
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
/bin/sh: tty: not found
/bin/sh: tty: not found
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'bash': Exec format error
env: can't execute 'perl': Exec format error
Failed to build tmp/.config-package.in
env: can't execute 'perl': Exec format error
make[1]: *** [/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:88: prepare-tmpinfo] Error 126
make: *** [/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:200: world] Error 2
root@LEDE:/tmp/sdk# make -j1 V=s menuconfig
/bin/sh: tty: not found
/bin/sh: tty: not found
make[1]: Entering directory '/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/scripts/config'
/bin/sh: cc: not found
make[1]: *** [<builtin>: conf.o] Error 127
make[1]: Leaving directory '/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/scripts/config'
make: *** [/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:103: scripts/config/mconf] Error 2
- 另外执行make  menuconfig -j1 V=s 错误如下：
> root@LEDE:/tmp/sdk# make  menuconfig -j1 V=s
> /bin/sh: tty: not found
> /bin/sh: tty: not found
make[1]: Entering directory '/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/scripts/config'
/bin/sh: cc: not found
make[1]: *** [<builtin>: conf.o] Error 127
make[1]: Leaving directory '/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/scripts/config'
make: *** [/tmp/lede-sdk-brcm2708-bcm2710_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:103: scripts/config/mconf] Error 2
- 不太理解是哪里出了问题了~求指点

aa65535, 2017-07-29:
这是系统有问题吧。

CXOldStar, 2017-07-30:
@aa65535 是指linux系统的问题还是SDK的问题？

aa65535, 2017-08-01:
@CXOldStar 系统问题，检查一下编译环境。

closed 2017-08-01 by aa65535